opencardev/crankshaft #1: Nothing happens when connecting phone

vb4life, 2018-02-25:
This is an awesome idea, but it doesn't seem to work for me.

I've flashed the image and can boot the RPi 3 with no problem. The screen comes up, waiting for a device to be connected. However, when I connect my phone (OnePlus 3), Android Auto starts on the phone, but nothing happens with the UI on the PI.

The log shows some errors, but nothing that tells me anything useful:
```
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.443027] [0x6d7d0310] [info]    [AaApp] [USBApp] Device connected.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.443862] [0x6d7d0310] [info]    [AaApp] [AndroidAutoEntity] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.445805] [0x72807000] [debug]   [AaApp] [AudioInput] create.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.446484] [0x72807000] [debug]   [AaApp] [AudioOutput] create.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.446867] [0x72807000] [debug]   [AaApp] [AudioOutput] create.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.447189] [0x72807000] [debug]   [AaApp] [AudioOutput] create.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.447493] [0x6d7d0310] [info]    [AaApp] [AudioInputService] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.454069] [0x6d7d0310] [info]    [AaApp] [AudioService] start, channel: MEDIA_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.454443] [0x6d7d0310] [info]    [AaApp] [AudioService] start, channel: SPEECH_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.454617] [0x6d7d0310] [info]    [AaApp] [AudioService] start, channel: SYSTEM_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.454774] [0x6d7d0310] [info]    [AaApp] [SensorService] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.454981] [0x6d7d0310] [info]    [AaApp] [VideoService] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.455171] [0x6d7d0310] [info]    [AaApp] [BluetoothService] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.455355] [0x6d7d0310] [info]    [AaApp] [InputService] start.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.571513] [0x6e7d0310] [info]    [AaApp] [AndroidAutoEntity] version response, version: 1.4, status: 0
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.571598] [0x6e7d0310] [info]    [AaApp] [AndroidAutoEntity] Begin handshake.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.587691] [0x6d7d0310] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 2282
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.651121] [0x6d7d0310] [info]    [AaApp] [AndroidAutoEntity] continue handshake.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.664682] [0x6efd0310] [info]    [AaApp] [AndroidAutoEntity] Handshake, size: 51
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.665141] [0x6efd0310] [info]    [AaApp] [AndroidAutoEntity] Auth completed.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692294] [0x6efd0310] [info]    [AaApp] [AndroidAutoEntity] Discovery request, device name: Android, brand: OnePlus ONEPLUS A3003
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692371] [0x6efd0310] [info]    [AaApp] [AudioInputService] fill features.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692423] [0x6efd0310] [info]    [AaApp] [AudioService] fill features, channel: MEDIA_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692458] [0x6efd0310] [info]    [AaApp] [AudioService] fill features, channel: SPEECH_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692484] [0x6efd0310] [info]    [AaApp] [AudioService] fill features, channel: SYSTEM_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692507] [0x6efd0310] [info]    [AaApp] [SensorService] fill features.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692546] [0x6efd0310] [info]    [AaApp] [VideoService] fill features.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692574] [0x6efd0310] [info]    [AaApp] [BluetoothService] fill features
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.692596] [0x6efd0310] [info]    [AaApp] [InputService] fill features.
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.760468] [0x6efd0310] [info]    [AaApp] [AndroidAutoEntity] requested audio focus, type: 4
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.760523] [0x6efd0310] [info]    [AaApp] [AndroidAutoEntity] audio focus state: 3
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.765763] [0x6efd0310] [info]    [AaApp] [AudioInputService] open request, priority: 0
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.765811] [0x6efd0310] [info]    [AaApp] [AudioInputService] open status: 0
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.771176] [0x6d7d0310] [info]    [AaApp] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.771245] [0x6d7d0310] [info]    [AaApp] [AudioService] open status: 0, channel: MEDIA_AUDIO
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.776517] [0x6d7d0310] [info]    [AaApp] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
Feb 25 19:57:40 raspberrypi autoapp[362]: [2018-02-25 19:57:40.776587] [0x6d7d0310] [info]    [AaApp] [AudioService] open status: 0, channel: SPEECH_AUDIO
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.781982] [0x6efd0310] [info]    [AaApp] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.782103] [0x6efd0310] [info]    [AaApp] [AudioService] open status: 0, channel: SYSTEM_AUDIO
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.789278] [0x6d7d0310] [info]    [AaApp] [SensorService] open request, priority: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.789327] [0x6d7d0310] [info]    [AaApp] [SensorService] open status: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.796448] [0x6d7d0310] [info]    [AaApp] [VideoService] open request, priority: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.796505] [0x6d7d0310] [info]    [AaApp] [OMXVideoOutput] open.
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.805836] [0x6d7d0310] [info]    [AaApp] [VideoService] open status: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.806322] [0x6e7d0310] [info]    [AaApp] [InputService] open request, priority: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.807010] [0x6e7d0310] [info]    [AaApp] [InputService] open status: 0
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.815449] [0x6d7d0310] [info]    [AaApp] [AudioService] setup request, channel: MEDIA_AUDIO, config index: 1
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.815511] [0x6d7d0310] [info]    [AaApp] [AudioService] setup status: 2, channel: MEDIA_AUDIO
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.821015] [0x6efd0310] [info]    [AaApp] [AudioService] setup request, channel: SPEECH_AUDIO, config index: 1
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.821073] [0x6efd0310] [info]    [AaApp] [AudioService] setup status: 2, channel: SPEECH_AUDIO
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.826355] [0x6d7d0310] [info]    [AaApp] [AudioService] setup request, channel: SYSTEM_AUDIO, config index: 1
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.826413] [0x6d7d0310] [info]    [AaApp] [AudioService] setup status: 2, channel: SYSTEM_AUDIO
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.832486] [0x6d7d0310] [info]    [AaApp] [SensorService] sensor start request, type: 13
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.837997] [0x6d7d0310] [info]    [AaApp] [VideoService] setup request, config index: 3
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.838050] [0x6d7d0310] [info]    [AaApp] [OMXVideoOutput] init, state: true
Feb 25 19:58:01 raspberrypi autoapp[362]: [2018-02-25 19:57:40.838362] [0x6d7d0310] [info]    [AaApp] [VideoService] setup status: 2
[... cut ...]
Feb 25 19:58:06 raspberrypi autoapp[362]: [2018-02-25 19:58:01.226239] [0x6d7d0310] [info]    [AaApp] [InputDevice] stop.
Feb 25 19:58:06 raspberrypi autoapp[362]: [2018-02-25 19:58:01.227407] [0x6d7d0310] [info]    [AaApp] [USBApp] Waiting for device...
Feb 25 19:58:06 raspberrypi autoapp[362]: [2018-02-25 19:58:01.227558] [0x6d7d0310] [debug]   [AaApp] [AndroidAutoEntity] destroy.
```

Any idea on what might be the issue?

I also tried compiling aasdk and openauto from source, but it is the same error..

htruong, 2018-02-25:
You should file a bug request with the openauto project :) I will keep it open to keep track of what devices work and what don't.

htruong, 2018-02-25:
Please provide with the Qt version you're trying to compile aasdk/openauto with, so we can debug this problem further. I will take care of this ticket.

vb4life, 2018-02-26:
I'm at work now, but I'll check when I get home.

But the first i tried was the crankshaft image downloaded on 2018-02-25.
Also, the compiled version was by following the instructions to compile on the same date.

RPi 3B, using a plain HDMI screen (I've ordered the official 7" touch screen, so will also test as soon as I get it)

OxygenOS 5.0.1
Android 8.0.0
Android Auto 3.0.570554-release

vb4life, 2018-02-26:
Actually, I just did a fresh install and compile of openauto etc, and it works. So it seems to be something in the crankshaft image...

htruong, 2018-02-26:
Thanks for reporting back @vb4life - I will figure out how to get an android phone that exhibits the problem and correct it in the next version.

vb4life, 2018-02-26:
Let me know if you need any logs, or if you want me to test anything...

or mute
the thread

vb4life, 2018-02-27:
I just recieved the 7" display and case, and after trying again with Crankshaft (same image), the phone connects just fine. So it might be an issue when using an external screen? My first test was with a HDMI -> DVI cable, so it might fail on the audio connection.

So you can probably close this issue, or modify it..

htruong, 2018-02-27:
Oh - that's right. I know it is doing something funky with the HDMI connection. HDMI doesn't work yet. So that mystery is solved. :)

parkerlreed, 2018-02-27:
Wait, you are connecting the PHONE via HDMI? Or do you mean the Pi is outputting HDMI and the phone conencted to the Pi via USB?

htruong, 2018-02-27:
vb4life used the HDMI touchscreen instead of the official DSI one. Bad title on my end.

parkerlreed, 2018-02-27:
Oh, weird that would cause an issue. I wonder if forcing an X11 build would work for the HDMI screen (for the time being)

vb4life, 2018-02-28:
I connected the PI to a normal (non-touch) screen. The screen also only had DVI connectors, so I used a HDMI -> DVI cable.

But using the official 7" screen works perfect, though.. :)

htruong, 2018-03-05:
For anyone having difficulties with custom HDMI screens, please try [Starting Crankshaft in X11 compatibility mode]( https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode#test-app-under-x11-compatibility-mode) on image 2018-03-05 or later.

htruong, 2018-03-07:
As of now with the introduction of the X11 compatibility mode, I think this should be resolved. Plus vb4life got their screen to work, so I'm closing this ticket down. Please feel free to comment and request to open a ticket if you feel this thread should continue.

closed 2018-03-07 by htruong