opencardev/crankshaft #2: List of devices that are known to work/not work

htruong, 2018-02-26:
Please comment here devices that are known to work and not work.

htruong, 2018-02-26:
Tested / Works:

- Google Pixel 1
- Raspberry Pi 3B, Official touchscreen.

htruong, 2018-02-26:
Tested / Does not work:

- One Plus 3
- ?

htruong, 2018-02-26:
Tested / Does not work:

- Adafruits HDMI screen - Flickers, wrong aspect ratio. Raspberry Pi does not supply enough power to it. It will not flicker if you plug it in through a hub.
- Adafruits AR1100 touch controller - Does not work at all.
- Image 2018-02-24.

kfechter, 2018-02-26:
Tested / Does not work:
Galaxy Note8 (N950U1)
Raspberry Pi 3B, Official touchscreen.
Image 2018-02-24.

Additional Information:
Android 7.1.1 (N950U1UEU3BRA5)/Android Auto 3.0.570554-release

When Connected to the raspberry pi, Android Auto opened and started the "new device" setup, but then showed an error screen with the message "Communication error 2, the phone and the car are running incompatible software"

I will test with the official open auto instructions and see if that works.

parkerlreed, 2018-02-26:
Tested/working (Linux x86_64/X11)
* Nexus 6P Oreo 8.1.0
* Desktop Linux (Thanks for the build instructions. Helped out a lot)
* Turned off the RPI3 build in the cmake flags and adjusted the paths to match my folders.

quarkpw, 2018-02-27:
Tested / partially works

Huawai P8 lite android 6 
android auto 3.0.570554
raspberry pi 3
Xenarc 7" hdmi touch screen.

Initial waiting for device screen is cropped
Phone connects ok, screen seems ok and working,
Touchscreen / usb mouse / usb keyboard not working.

VBMCBoy, 2018-02-27:
Tested / partially works

* LG G4 H815 TWN
* Raspberry Pi 3B, official Touchscreen
* Image 2018-02-24

bluetooth does not work yet

parkerlreed, 2018-02-27:
Problem with Bluetooth is I think it depends on the HFP (hands free) profile working in Pulseaudio. I've been messing with it on my x86_64 testbed and while I do get a connection with Ofono running, no audio comes across from the phone.

jlukanc1, 2018-02-27:
Working (so far):
Droid Turbo 2
Raspberry Pi 3B, Official touchscreen
Image 2018-02-24

zt4, 2018-03-01:
Tested / Works:

Google Pixel 2 XL, Android Oreo 8.1
Raspberry Pi 3B, Official touchscreen.
Image 2018-02-28.

prakashdivyy, 2018-03-02:
Tested / Works:
- Xiaomi Mi A1, Android Oreo 8.0.0
- Raspberry Pi 3B, Official touchscreen.
- Image 2018-02-24

prakashdivyy, 2018-03-02:
Tested / Works:

Xiaomi Mi A1, Android Oreo 8.0.0
Raspberry Pi 3B, Official touchscreen.
Image 2018-02-28

Tibbey, 2018-03-03:
Tested / Works:

Nexus 6, Android Nougat 7.1.1
Raspberry Pi 3B, Official touchscreen
Image 2018-03-03

nacbooth, 2018-03-03:
Tested / Does not work

    Motorola G5S, Android 7.1.1
    Pi  3 model B / Official 7" screen.
    Image: crankshaft-2018-03-02.zip.
    
    Nothing happens at all when plugged in to USB (was same for the previous two versions as well)

tkraspi, 2018-03-03:
Xiaomi A1 Android Oreo 8.0.0
Raspberry Pi 3 B, Official Raspberry Pi LCD  7 inch and 
Waveshare LCD 10 inch HDMI (D) Touchscreen 
Image 2008-02-2018
*) for Waveshare LCD no sound over LCD speaker

htruong, 2018-03-03:
@nacbooth Just wanted to make sure, is it the G5S Plus? I am considering taking the plunge and buying one to test.

joufflu1, 2018-03-03:
Tested / Works
Huawey P8 Lite 2017 (android 7.0 EMUI 5.0.1)
Raspberry Pi 3 B + Official screen
Image 2018-03-02

nacbooth, 2018-03-03:
@htruong - it's not the plus variant:  Motorola Moto G5S, 32 GB, (Single Sim), 3GB RAM, UK, SIM-Free

nacbooth, 2018-03-03:
Updated:  **Changed USB cable**, and now it looks like things work.  Need to do further on-road tests, but looks OK in the workshop.

Motorola G5S, Android 7.1.1
Pi  3 model B / Official 7" screen.
Image: crankshaft-2018-03-02.zip.

htruong, 2018-03-03:
@nacbooth That is indeed incredible. I am very happy to hear! I'd have to suggest it in the top post.

tkraspi, 2018-03-04:
Tested / Works
Mi A1 Oreo 8.0.0
Pi 3 model B , Waveshare LCD 10 inch HDMI (D), touchscreen
Crankshaft-2018-03-02
for increase volume over LCD speakers : 
enable HDMI audio over raspi-config and
add line in /etc/pulse/default.pa
# Set volume to 50% on boot
set-sink-volume 0 32768

keredson, 2018-03-04:
> Updated: Changed USB cable

@nacbooth we had a similar issue in my wife's car w/ the real android auto.  the usb in question worked fine elsewhere, but not in her car.  but a different one did.

htruong, 2018-03-04:
FWIW, Someone commented on my youtube video, 
Gabriele Amodeo - "Working great with Samsung Galaxy S7 and Waveshare 7" inch LCD Touch, a little mod on config.txt is required but everything is good!"

derRAV3N, 2018-03-04:
Tested/Not Working:

Pixel 2 & Galaxy S4
Raspberry Pi 3 B, External 24" HDMI Screen.
Newest crankshaft build.

Probably because of Issue #1 not working, also I don't have an USB -> USB-C cable available at the moment, so I tried it with an USB -> Micro-USB cable with an Micro-USB -> USB-C Adapter attached.

Booting fine, Smartphone recognized and showing the Android Auto Splashscreen, but the HDMI screen is showing the default unconnected screen of Crankshaft.

nwithan8, 2018-03-04:
Tested/Works

Essential Phone and Nexus 6P
Raspberry Pi 2, official 7" touchscreen
Crankshaft-2018-03-02

https://youtu.be/U3KxJ0raolU

keredson, 2018-03-05:
it works fine w/ the [GeeekPi 7 Inch 1024x600 Capacitive Touch Screen](https://www.amazon.com/gp/product/B075QCXLPF/ref=oh_aui_detailpage_o02_s01?ie=UTF8&psc=1)
if you do their `config.txt` adjustment:
```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 3 0 0 0
hdmi_force_hotplug=1
```

but i don't recommend it, as android auto only supports 480p*, and it looks fuzzy scaled to 600p.

(*) or 720p or 1080p if in dev mode, neither of which look any better at 600p.

blweldon2, 2018-03-05:
Tested / Works

- Sony Xperia XZ Premium / Android 8.0 Oreo
- Raspberry Pi 3B / official 7-inch touchscreen
- Image 2018-03-02

Had to add lcd_rotate=2 to the config file to correct my display orientation with my case, but everything works as flawlessly as can be expected.

htruong, 2018-03-05:
For people using custom HDMI screens, please try [Starting Crankshaft in X11 compatibility mode]( https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode#test-app-under-x11-compatibility-mode) on image 2018-03-05 or later.

tehniemer, 2018-03-05:
@htruong is there any hope for the Adafruit AR1100 touch controller or other touch controller in the near future? I'm hoping to add a touch overlay and HDMI to RCA converter to my Pontiac G8 factory radio for a really clean installation,

htruong, 2018-03-05:
@tehniemer I no longer have the ar110 touch controller to test. Do you have one available to test with X11 mode?

tehniemer, 2018-03-05:
@htruong not currently, it is on my list of things to purchase though.

dnnadmins, 2018-03-05:
Samsung Galaxy S8 + Crankshaft v0.1.6 works fine.

tehniemer, 2018-03-06:
Tested / Works (partially)

- LG Nexus 5 / Android 7.1.2
- Dell U3415W (no touchscreen)
- Image 0.1.6

The display works fine, but since the screen is not touch enabled I cannot confirm full functionality.

ubergeekseven, 2018-03-06:
@kfechter so you did get it to work with the note 8? What did you have to do?

floyduww, 2018-03-06:
Tested / partially works

-  Pixel 1
- pi 3
- image 2018-03-05
- [ebay 7 Inch TFT LCD Touchscreen + Driver HDMI VGA w/ backup camera](https://www.ebay.com/itm/US-Ship-7-Inch-TFT-LCD-Touchscreen-Driver-HDMI-VGA-for-Raspberry-Pi-3-Model-B-/121762136526)

Touchscreen is eGalax.   And the y-axis is flipped and setting the inverty option causes autoapp to crash.

kfechter, 2018-03-07:
@Xseam @ubergeekseven  I ended up installing normal raspian and compiling the aasdk and OpenAuto projects manually. everything seemed to work after that

reverendbacon, 2018-03-08:
Tested / Works:

Google Pixel 2, Android 8.1.0
Raspberry Pi 3B, Official touchscreen.
Image 0.1.6 2018-03-05

EveAeternam, 2018-03-08:
Tested / Works:

* Samsung Galaxy S7 Edge, Android 7.0
* Raspberry Pi 3B, Official Touchscreen v1. 0
* Image 0.16 (2018-03-05)

rcrichton, 2018-03-08:
Tested / Works:

* OnePlus 5T, Android 8.0
* Raspberry Pi 3B, Official touchscreen.
* Image 2018-03-05.

GerRudi, 2018-03-08:
Tested / Works: 

OnePlus 2, Android 7.1.2
OS: LineageOS 14.1, Build from 10/2017
Kernel (Phone): Boeffla-Kernel-6.1-beta8 (custom built with alternative charging drivers)
Raspberry Pi 3B, Official touchscreen.
Image 0.1.6 (2018-03-05)

cowmix, 2018-03-10:
Tested / Works:

MotoZPlay, Android 7.0
Raspberry Pi 3B, Official Touchscreen v1. 0
Image 0.16 (2018-03-05)

bhwithun, 2018-03-11:
Tested / Not Working:

> Google Pixel 1 XL, Android Oreo 8.1
> Raspberry Pi 3B, Official touchscreen.
> *Image 2018-02-28*

> Google Pixel 1 XL, Android Oreo 8.1
> Raspberry Pi 3B, Official touchscreen.
> *Image 2018-03-10*

Both scenarios produce the same result.  The Pi boots and gets the the "waiting" screen.
(NOTE: the 03-10 has a *very* different waiting screen)
When the Pixel is plugged in, the Pixel reacts, showing "Welcome to Android Auto", offering two choices:
* NO THANKS
* GET STARTED

On either build, when I choose "GET STARTED", the Pixel returns to its desktop screen.  Nothing else happens.  The Pi does not react at all when the phone is connected.

I have observed that while connected, my Pixel assigns the "Charging this device" role to the USB connection.  I have tried to change that role to each of the following:
* Charge this device
* Supply power
* Transfer files
* Transfer photos (PTP)
* Use device as MIDI

When I choose "Transfer photos (PTP)" I am greeted once again with the "Welcome to Android Auto" screen.  My choice to "GET STARTED" kicks me back to the Pixel desktop and the Pixel has returned to "Charging this device" role.

htruong, 2018-03-11:
@bhwithun Do you mind trying with another USB cable? The Pixel has to be one of the most thoroughly tested devices - I have a Pixel 1 myself. The "plug phone in" screen is OK, it changed quite a bit between version.

measle, 2018-03-11:
TESTED WORKING!
LG G4 Running H-Rom V20 PORT V2.2  (Android 7.0))
Kernel version 3.10.84
Android security patch level July 1,2017

Raspberry Pi 3B
Raspberry Pi 3B, Official touchscreen.
Image 2018-03-05

Everything working flawlessly; THANK YOU!

thanhtantran, 2018-03-11:
Tested work with
- RPi 3
- Waveshare 5inch screen (must enter dev mode to edit screen resolution and touch driver)
- Moto Z Play (XT1635-02)

bhwithun, 2018-03-11:
## Re-Tested

I tried the Pixel 1 again.  Same results.  I have tried 2 different USB C cables.  The same cables that do not work with my Pixel 1 XL work fine with my Pixel 2.

Plugging in the Pixel 2 has worked 11 out of 11 times.
Plugging in the Pixel 1 has worked 0 out of 15 times.

# Drive Test
I did a brief couple of road tests today.
> Raspberry Pi 3 - Model B - ARMv8 with 1G RAM (powered by dedicated RavPower Turbo-Series 2.4A)
> Pi Foundation Display - 7" Touchscreen Display for Raspberry Pi (powered by dedicated Anker IQ)
> Jabra 510 Speaker (USB connected/powered by RPi)
> Google Pixel 2 (8.1.0)

My RPi was showing the "lightning bolt" icon almost constantly.  I'm not sure why it would be experiencing underpower, but apparently it was.

### Navigation Only
The navigation worked fine, the audio also worked fine.  No issues.

### Music Only
Listening to music worked fine, no issues.

### Music + Nav + Map
After selecting a destination and beginning navigation, I chose some music to listen to.  The music began fine, but it was pretty clear that the audio was affected by the graphical navigation.  When the map is zooming or panning, the audio gets choppy.

When the navigation voice begins speaking, the music audio gets considerably worse.   After making one left turn, the music remained choppy and laggy (it was painfully unlistenable) for almost a full minute.  It did eventually recover and sound like music again after that minute.

### Music + Nav
After a while I changed the display so that I was no longer looking at the map.  I chose the music screen.  The audio and the navigator voice worked much better this way.  Maybe the RPi just doesn't have enough power to do music + navigation + maps.

I noticed that the display changed song titles about 1 minute before the song ended.  I don't think that's normal.  I think the on-screen song title is supposed to always reflect exactly what I'm listening to.

While I was looking at the music screen I did notice that the periodic instructions from the navigator would produce a "pop-up" banner at the top of the screen, telling me about the upcoming turn.  These pop-ups were very well coordinated with the navigator voice. 

#### Music Fade
I noticed that the music would fade periodically.  This sounded like the decreased-volume that would be associated with navigation voice saying something, but the music fade was not coordinated with the navigators voice.  Sometimes it was, but usually it was not.  Sometimes the navigator spoke while the music was full volume, and sometimes the navigator spoke while the music volume was decreased.

### End-of-Trip
After reaching my destination I noticed that the music still faded (lowered its volume) periodically.  The navigation was certainly not active at this time, so I'm not sure why the music volume would be reduced at all, but it continued to do so.  Maybe that lightning bolt on the screen is relevant.  Maybe my system is chronically underpowered.

I disconnected the Pixel 2 and the display returned to the "waiting for connection" screen (3/10/2018 build).  I pushed the power icon.  I think this normally shuts down the system.  This time it did not shutdown.  Instead, I was shown the terminal and all the "[  OK  ] Started LSB: Switch to ondemand cpu governor ..." kinds of messages.  I seem to recall seeing these before as the system was shutting down, but this time they stayed, along with these other messages

> [    76.951557] Under-voltage detected! (0x00050005)
> [    83.191573] Under-voltage detected! (0x00050005)
> [  112.311574] Under-voltage detected! (0x00050005)
> [  126.871557] Under-voltage detected! (0x00050005)

Mearman, 2018-03-12:
- Pixel 2 XL
- - Android 8.1.0
- Official 7" Touchscreen
- Raspberry Pi 3B
- Versions:
- - 0.1.6 2018-03-05: ✓
- - 0.1.7 2018-03-10: ✓

htruong, 2018-03-13:
@bhwithun  The audio should no longer jank with the latest release :) Enjoy

bhwithun, 2018-03-15:
## Tested / Works

> Pixel 2 / Android 8.1.0
> 0.2.0 / 2018-03-13
> Raspberry Pi 3B
> Official 7" Touchscreen

## Tested / Not Working

> Pixel 1 XL / Android 8.1.0
> 0.2.0 / 2018-03-13
> Raspberry Pi 3B
> Official 7" Touchscreen

Developer says that this works with Pixel 1 XL, and I'm inclined to believe that, but it's not working with mine.  Are there steps I can take to troubleshoot this?  I have un/reinstalled Android Auto.  AA actually works fine on my phone as a stand-alone, but when connected to the Pi, it's not working.  The Pi doesn't acknowledge that a Pixel 1 XL has been plugged in.

When I unplug the Pixel 1 XL and plug in a Pixel 2 it works perfectly.  (The music/nav voice problems seem to be gone).  I trust the cable, the Pi, the Screen, the power source.  The one variable I cannot change is my Pixel 1 XL.  I only have one of them.  It's nothing weird.  Not rooted.  Not hacked.  Not jailbroken  -- if those terms even apply to the Pixel phones.

Here's a pic for fun.  I've drawn the USB cable in to show how that it's connected.

The Raspberry Pi is flat-out ignoring my Pixel 1 XL.

[image]

andreums, 2018-03-15:
Tested with Makibes 7 inch touch Screen ( https://www.amazon.co.uk/Makibes-Capacitive-1024X600-Resolution-Raspberry/dp/B01AW8POBE/ref=sr_1_1_sspa?ie=UTF8&qid=1521136451&sr=8-1-spons&keywords=makibes+7+inch+capacitive+touch+screen&psc=1 ) and Leagoo S8.

Just followed this tutorial: https://www.youtube.com/watch?v=Q9gHsR4foLM with the 13th of March release.
Only issue is that I can't get voice by bluetooth speaker or by my phone.
I can say 'Ok Google' and shows speech recognizer, but not recognizes my voice/anything.

Thank you for your efforts!
Greets from Spain.

jlukanc1, 2018-03-16:
Tested/working
LG V30 (included usb-c cord)
Pi 3b
Version alpha0.2.0 2018-03-13
w/ Official display

spaceyleo, 2018-03-19:
Tested without touchscreen: Working
One Plus 3 (USB C cable from amazon - not official)
Pi 3b+
Version: 2018-03-13
Display: Lg 23in HDMI monitor no touchscreen
Ok google works.
Didn't try bluetooth.

edinn, 2018-03-22:
Tested / Works

LG Nexus 5x / Android 8.1.0
HP Pavilion 23xi display
Image 0.2.0

chaoman, 2018-03-22:
Tested working:

HTC 10 Android Oreo 8.0
Raspberry Pi 3B, Official touchscreen.
Image 2018-03-13.

Tested not fully working but with workaround:
Sound Blaster Play! 2 external USB sound card. Filed [https://github.com/htruong/crankshaft/issues/50](url)

ryanlovett, 2018-03-23:
Phones: Pixel XL 1, Nexus 5
Raspberry Pi 2B
Raspbian Stretch Lite, manually compiled aasdk and openauto.
CannaKit power adapter

Initially neither phone worked. After connecting either, openauto would not respond in any way. `dmesg` would show that the Pis detected both phones, but the app would only sit waiting. I tried stopping Android Auto on the Pixel, deleting its cache and data, then reinstalling, but that had no effect. I also tried plugging each phone into a USB hub and the hub into the Pi.

I ran `rpi-update` to pull the latest firmware, and everything worked after the reboot.

Maju3, 2018-03-28:
Tested/Working
Phone: Asus Zenfone 3 Laser
Android:7.1
Raspberry Pi 3B
Image:2018-03-13
Custom Touchscreen: CTF 800 wmsl (www.cartft.com)
USB DAC (soundcard):Hiresfi usb dac 5122A

To get the HDMI Output to 800x480 vga Resolution I have edited config.txt:

disable_overscan=1
hdmi_force_hotplug=1
hdmi_group=2

hdmi_mode=87

#set specivic CVT mode

hdmi_cvt 800 480 60 6 0 0 0
config_hdmi_boost=4

krizangrandi, 2018-03-29:
Tested/Working
Phone: Xiaomi redmii Note 4X
Rom: Pixel Experience
Android: 8.1
Raspberry Pi 3B
Image: 2018-03-13
Generic HDMI Display 6.5" (V59.AV2 controller + AT065TN14 + touch panel)
Touch: eGalax generic controller

To get the HDMI Output to 800x480 I have edited config.txt:

hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_drive=1
hdmi_cvt 800 480 60

I had the y flip touch problem, i resolved using X11 mode and xinput_calibrator on raspian and coping the calibration result.
It's not necessary using: option "InvertY" "1"
And now i have a better precision input offline the touch.

This is my 99-configuration.conf :

Section "InputClass"
Identifier "calibration"
MatchProduct "eGalax Inc. USB TouchController"
Optional "calibration" "80 1960 1880 160"
EndSection

tehniemer, 2018-03-30:
Tested/Working

Phone: Nexus 5
Android: Lineage 14.1
Raspberry Pi 3B
Image: 0.2.0
Custom Touchscreen: Pontiac G8 factory infotainment (RCA no HDMI), touch overlay, Adafruit AR1100 touch controller (eGalax also works)

X11 mode w/ xinput_calibrator on raspbian and copying the calibration result.

ask77, 2018-03-31:
Tested/working
Phone: OnePlus 5T running Oreo8.1 
Raspberry Pi 3B
Custom 1024 X 600 Touchscreen from Aliexpress running from hdmi port with the config.txt file modified wit the following text
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1

s0len0id, 2018-04-04:
Tested /Works:

* Motorola Z2Play (XT1710-09/Android 7.1.1)
* RPI 3B, Playstation Eye (for the mic array), Samsung TV P2770HD in X11mode
* crankshaft-2018-04-01-1.zip / Android Auto 3.1.581053

btw: the Playstation Eye might be a very cheap but excellent noisecancelling mic according to:
https://medium.com/snips-ai/benchmarking-microphone-arrays-respeaker-conexant-microsemi-acuedge-matrix-creator-minidsp-950de8876fda
(and in the future maybe a dashcam too? ;)

[116 more replies]