opencardev/crankshaft #4: Figure out a sensible way to turn on/off the system

htruong, 2018-02-26:
Jank out the USB cable is dumb and dangerous. Figure out a way to do it more gently.

htruong, 2018-02-28:
The issue is partially resolved with read-only fs in alpha 0.1.1-2018-02-28. Now there is no more corruption, but users still have to yank the cables to turn the system off.

parkerlreed, 2018-02-28:
Maybe suggest to OpenAuto to add a poweroff item to the main menu?

htruong, 2018-02-28:
@parkerlreed  I think the general idea here is "do it and do a pull request, and hope he accepts it" instead of "asking f1xpl to do it for me."

parkerlreed, 2018-02-28:
That's what I meant. :smile: Could have worded that better.

htruong, 2018-03-03:
This is now resolved with image 2018-03-02. Now users can turn off the system with the power button on the main UI. Users can also temporarily put it to sleep (disable backlight), so it doesn't look totally ridiculous when the car is parking somewhere.

closed 2018-03-03 by htruong