opencardev/crankshaft #5: Error -2 with several phones

htruong, 2018-02-26:
Some phones were reported to have problems connecting to OpenAuto. The problem goes away with official OpenAuto on X11. Figure out a way to get around the problem.

borconi, 2018-03-09:
I haven't tested crankshaft myself yet but this I saw error 2 in case the headunit part is set to 720p or 1080p and the dev settings on the phone are not enabled for it. 

Other popular reason for error 2 is bad communication between the devices as a first try a different usb cable and/or try a different power adapter, fluctuating power or underpowered USB can cause this!!!!

jjpython, 2018-03-24:
i get error 2 on LG G5se, Tried different cable, Using an oficial power supply for the raspberry pi3, tried different screen settings as well

htruong, 2018-03-24:
@jjpython Someone tipped off this observation: https://www.reddit.com/r/crankshaft/comments/86u7xl/error_2_power_issue/

There is something about power supply or USB cable that caused that.

htruong, 2018-04-26:
If no one has further comments on this issue before 04/04, I will close this issue as "resolved" :)

closed 2018-05-08 by htruong

Archer1632, 2018-05-11:
I just built last night and was getting error2 right off the bat.  It was because I had not enabled Android Auto Dev mode and enabled 720p and 1080p mode.  If an unsupported video mode is on crankshaft but not on the personal device then it errors out.  Easy fix, for me at least.  Galaxy Note 8, Raspberry Pi 3 B+, Version 0.2.3 2018-04-18