opencardev/crankshaft #6: Package as Debian packages

htruong, 2018-02-26:
It would be nice to be able to distribute OpenAuto/Crankshaft as a Debian package. Figure out how.

parkerlreed, 2018-02-26:
Do you have generic building instructions for `aasdk`/`openauto` for Linux? I've been attempting on my regular computer (had to manually copy the `aadsk` include/ directory to my /usr/include) and got `openauto` to almost compile but ended up with a bunch of undefined references when linking `autoapp`.

parkerlreed, 2018-02-26:
Success! 
[image]

hawkeyexp, 2018-07-09:
I think should be closed cause deb package is not the way to create a whole system :-)

closed 2018-07-12 by hawkeyexp