opencardev/crankshaft #10: Smooth upgrade

htruong, 2018-03-01:
In the current state of Crankshaft, every upgrade needs to be done by reflashing the SD card. A better way to upgrade would be either OTA or at least through a USB drive/Android phone.

htruong, 2018-03-01:
Depends on #8.

halflifestudios, 2018-04-04:
Is there a remote or USB way to update the builds? My device is screwed into the enclosure which makes it difficult to get access to the SD card everytime to upgrade. An easier or remote way would be ideal in order to keep up with your great updates. The software has been working great and have been using it every day. Sometimes my Note 8 doesn't recognize right away and I have to reboot the OS, but other than that it has been doing very well. Thank you again for all your work on this endeavor.

hawkeyexp, 2018-06-27:
With upcoming builds based on crankshaft-ng branch system will get an option to update system by attach usb stick with update file :-)

closed 2018-07-09 by hawkeyexp