opencardev/crankshaft #11: Choppy audio

htruong, 2018-03-01:
- Phone: Pixel 1 - Android Oreo 8.1
- Image 2018-02-28

The audio is choppy when navigation is used at the same time as podcast playback. It happens in OpenAuto too. Find a way to mitigate the problem.

parkerlreed, 2018-03-01:
Run into the same with my testing on regular Linux. The Pulseaudio config edits kinda heloed but didn't go away completely. You can hear it here.

https://youtu.be/POybXgob7nk?t=70

htruong, 2018-03-03:
This problem is resolved in Crankshaft alpha 0.1.5 2018-03-02.

Reasons: The raspberry pi CPU had to do speex-float which it chokes itself. Changing pulseaudio config to speex-fixed-1 helps a lot. Now it still stutters when it needs to draw a lot, but this is not related to pulseaudio anymore. Closing ticket.

closed 2018-03-03 by htruong

parkerlreed, 2018-03-03:
For reference

pulseaudio `/etc/pulse/daemon.conf`

````
resample-method=speex-fixed-1
````

floyduww, 2018-03-13:
try to remove tsched=0 from default.pa   that helped my setup