opencardev/crankshaft #9: Bluetooth A2DP

htruong, 2018-03-01:
It is possible to provide A2DP with the RPi3's Bluetooth chip. Figure out a way to allow users to connect to their stereo. Probably not going to happen before the overlay problem is solved.

borconi, 2018-03-09:
Problem with A2DP is that RPi3 build in bluetooth will be able to work correctly in A2DP profile but it will fail for HFP (Ofono + Pulseaudio + Bluez5) combination. Also Android Auto will automatically disable the A2DP on the phone when it connects to a unit which provides a bluetooth address. Therefore if you want to have A2DP capabilities on your phone, you must completely remove the bluetooth declaration, sending a empty mac address will make the phone go crazy so it's not the solution. 

I'm not a C guy but if you want A2DP to work with an external device and you also want handsfree to work with an external device, then you need to remove the ```serviceList.emplace_back(this->createBluetoothService(messenger));``` from https://github.com/f1xpl/openauto/blob/development/src/autoapp/Projection/ServiceFactory.cpp

If there is no bluetooth channel defined when the connection is initiated, the phone will continue to use any bluetooth devices as normal. In other words if you have a hfp in the car already wich can play a2dp as well, you can use that just as before in this case the rpi3 will become an external screen for your phone nothing more nothing less.

Hope it makes sense.

htruong, 2018-03-09:
Thanks for the wonderful tip @borconi, I think it will make it easier for many users to have the ability to configure. Having that information in hand will definitely help with the ability to make that configuration option possible.

parkerlreed, 2018-03-09:
Umm so does that Bluetooth line change also affect if the phone isn't even connected to the computer running OpenAuto?

Bluetooth from phone to computer was NOT connected. Bluetooth on phone was connected to my car (Media and Phone profile)

Plugging in USB the Phone and Media stay connected but Android Auto seemingly forces Media audio through USB.

borconi, 2018-03-09:
@parkerlreed - Thanks for reminding me about that... totally forgot. @htruong removing the bluetooth config on it won't suffice (sorry I totally forgot about that), you also need to remove the media channel declarations as well if you want AA to output sound on phone speaker/bluetooth a2dp.

As long as media channels are defined, audio will be routed to USB.

parkerlreed, 2018-03-09:
@borconi Mind making a patch for OpenAuto to test that out? Would be kinda cool to just have OpenAuto be a "controller" but not handle any audio itself (One benefit being better handling of the media button in some cars)

tehniemer, 2018-03-09:
This would need to be configurable somehow for those cars that don't have
stereo bluetooth, or bluetooth at all.

parkerlreed, 2018-03-09:
Fully agree. I meant just a patch to test locally, not actually changing any behavior in master.

htruong, 2018-03-10:
@borconi Have you able to get this to work with your implementation? I am using aasdk and tried to disable all audio services, and android auto app didn't like that at all (error -2). I can ask Android Auto to leave my mediaAudioOutput and speechAudioOutput streams alone and let the phone handle it, but can't seem to disable audioInput and systemAudioOutput.

```C++

ServiceList ServiceFactory::create(aasdk::messenger::IMessenger::Pointer messenger)
{
    ServiceList serviceList;

    IAudioInput::Pointer audioInput(new AudioInput(1, 16, 16000), std::bind(&QObject::deleteLater, std::placeholders::_1));

    IAudioOutput::Pointer mediaAudioOutput(new AudioOutput(2, 16, 48000), std::bind(&QObject::deleteLater, std::placeholders::_1));

    IAudioOutput::Pointer speechAudioOutput(new AudioOutput(1, 16, 16000), std::bind(&QObject::deleteLater, std::placeholders::_1));

    IAudioOutput::Pointer systemAudioOutput(new AudioOutput(1, 16, 16000), std::bind(&QObject::deleteLater, std::placeholders::_1));

    serviceList.emplace_back(std::make_shared<AudioInputService>(ioService_, messenger, std::move(audioInput)));
    serviceList.emplace_back(std::make_shared<SystemAudioService>(ioService_, messenger, std::move(systemAudioOutput)));

	// disabling media and speech works fine, but disabling audioInput and systemAudioOutput makes android auto client upset
    if (0) {
        serviceList.emplace_back(std::make_shared<MediaAudioService>(ioService_, messenger, std::move(mediaAudioOutput)));
        serviceList.emplace_back(std::make_shared<SpeechAudioService>(ioService_, messenger, std::move(speechAudioOutput)));
    }

    serviceList.emplace_back(std::make_shared<SensorService>(ioService_, messenger));
    serviceList.emplace_back(this->createVideoService(messenger));

    serviceList.emplace_back(this->createBluetoothService(messenger));

    serviceList.emplace_back(this->createInputService(messenger));

    return serviceList;
}
```

borconi, 2018-03-10:
@parkerlreed - Sorry I'm not good with the C family so it will be quite difficult for me to create a test build.

@htruong 
You need to leave the audio input enabled, that cannot be disabled, it will cause communication error 2 or Google Play Service crash on phone. I'm still old school on this connection part, haven't switched over to protobuf yet, but this is how I do it, I have a bytearray (chararray): 

```
byte sd_buf2 []={0x00, 0x06, 
//Touch+input CHANNEL
0x0A, 34,   0x08, 0x01, 0x22, 30, 0x0A, 20, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x54, 0x55, 0x56, 0x57, 0x58, 0x59, 0x5A, 0x12, 0x06, 0x08, 0x80, 0x0f, 0x10, 0xb8, 0x08,
//0x0A, 0x0F, 0x08, 0x01, 0x22, 0x0B, 0x0A, 0x01, 0x54, 0x12, 0x06, 0x08, 0xA0, 0x06, 0x10, 0xE0, 0x03, 
//0x0a, 0x12, 0x08, 0x01, 0x22, 0x0e, 0x0a, 0x0c, 0x01, 0x02, 0x04, 0x13, 0x14, 0x15, 0x16, 0x17, 0x54, 0x80, 0x80, 0x04,

// SENSOR CHANNEL
0x0A, 16, 0x08, 0x02, 0x12, 12, 0x0A, 0x02, 0x08, 0x01, 0x0A, 0x02, 0x08, 0x0A, 0x0A, 0x02, 0x08, 13, 
//                       18,  12,     10,    2,    8,    1,  10,   2,     8,    10, 10,    2,    8,    13
//General Notification channel currently not working I might be doing something wrong...
//0x0a, 4,     0x08, 13,0x6A,0,

// Video CHANNEL
0x0A, 0x15, 0x08, 0x03, 0x1A, 0x11, 0x08, 0x03, 0x22, 0x0D, 0x08, 0x01, 0x10, 0x02, 0x18, 0x00, 0x20, 0x00, 0x28, -96, 0x01, 0x30, 0x00, 
// MIC CHANNEL
0x0A, 4+4+7,0x08, AA_CH_MIC, 0x2A, 4+7, 0x08, 1,    0x12, 7,    0x08, -128, 0x7d, 0x10, 0x10, 0x18, 1,
//0x0A, 29,   0x08,     8,      0x32, 25, 0x0A, 17, '0','0',':','0','0',':','0','0',':','0','0',':','0','0',':','0','0',0x12,1,4,0x1A,1,1,
//MediaPlaybackService?????
0x0a, 6, 0x08, 9,0x4A,2,0x08,01,
//Navigation Status Service (below code looks to be working, we are receiveing 2 messags, 8004 and 8005, 8005 is probably next turn distance/time and 8004 is the graphical part, inc street name...
0x0a, 17, 0x08,10,0x42,0x0d,0x08,-84,0x02,0x10,0x01,0x1a,0x06,0x08,0x40,0x10,0x40,0x18,0x10,
//0x0a, 6, 0x08,10,0x42,2,0x08,01,
//0x0a, 6, 0x08, 9,0x7A,2,0x08,01,

// CAR DEFINITION
						
                        0x12, 0, //'H','e','a','d','u','n','i','t',//1, 'A', // Car Manuf          Part of "remembered car"
                        0x1A, 8, 'R','e','l','o','a','d','e','d',//1, 'B', // Car Model
                        0x22, 4, '2', '0', '1', '7',//1, 'C', // Car Year           Part of "remembered car"
                        0x2A, 4, '0', '0', '0', '1',//1, 'D', // Car Serial     Not Part of "remembered car" ??     (vehicleId=null)
                        0x30, 0,//0,      // driverPosition
						
                        0x3A, 8, 'H','e','a','d','u','n','i','t',//1, 'E', // HU  Make / Manuf
                        0x42, 4, 'H', 'U', '1', '5',//1, 'F', // HU  Model
                        0x4A, 4, 'S', 'W', 'B', '1',//1, 'G', // HU  SoftwareBuild
                        0x52, 4, 'S', 'W', 'V', '1',//1, 'H', // HU  SoftwareVersion
                        0x58, 1,//1,//1,//0,//1,       // ? bool (or int )    canPlayNativeMediaDuringVr
                        0x60, 0,//1,//0,//0,//1        // mHideProjectedClock     1 = True = Hide	*/	
						
//THE LAST 3 CHANNEL KEEP THEM HERE SO WE CAN EXCLUDE THEM IF NEED BE.
//WifiProjectionService (Phone BSSID????)
//0x0a, 4+16,     0x08, 10,0x72,16,0x0A,'H','e','a','d','u','n','i','t',' ','R','e','l','o','a','d',

0x0A, 4+6+7, 0x08, AA_CH_AU2, 0x1A, 6+7, 0x08, 1,  0x10, 2, 0x1A, 7, 0x08, -128, 0x7d,         0x10, 0x10,   0x18, 1,
0x0A, 4+6+7, 0x08, AA_CH_AU1, 0x1A, 6+7, 0x08, 1,  0x10, 1, 0x1A, 7, 0x08, -128, 0x7d,         0x10, 0x10,   0x18, 1,
0x0A, 4+6+8, 0x08, AA_CH_AUD, 0x1A, 6+8, 0x08, 1,  0x10, 3, 0x1A, 8, 0x08, -128,   -9, 0x02,   0x10, 0x10,   0x18, 02,

};
```

As you see from the comment I'm just trimming the end of the bytearray in case I do not want to sink the audio. The mic channel is my audio input channel. Hope it helps.

htruong, 2018-03-10:
I have created a pull request with @f1xpl https://github.com/f1xpl/openauto/pull/38

htruong, 2018-03-10:
For anyone following this thread, I have pushed an experimental build, in which OpenAuto will not take over the phone's Media and Speech output (so the phone continues outputting by whatever means it normally does, bluetooth, headphone jack, etc). 

Download at: https://github.com/htruong/openauto/releases/tag/v0.0.1-aud-exp

htruong, 2018-03-13:
Fixed as of 0.2.0! Thanks @borconi !

https://www.reddit.com/r/crankshaft/comments/842r3g/breakthrough_release_crankshaft_alpha_020_and

closed 2018-03-13 by htruong