opencardev/crankshaft #12: Cannot enable bluetooth

strabbit, 2018-03-02:
When I enter "Settings" and select the "Bluetooth" tab, and select the "Use local adapter" radio option and then save settings, the settings app closes. I briefly see the terminal prompt, and then I see "Waiting for device" again.

If I enter settings, my change doesn't seem to be persisted, and "Disable bluetooth" is selected.

- What phone do you have? What OS version?

Google Pixel (2016)
Oreo (8.1.0)

- Does it happen with OpenAuto on X11 too, or you don't know?

Don't know

- Did you try to compile OpenAuto with Xorg instead of EGL? Did you follow the build instructions on https://github.com/f1xpl/openauto/wiki/Build-instructions? What is your libQt version and does that come with raspbian?

No. Nope. Whatever version comes with Crankshaft 2018-08-28

htruong, 2018-03-02:
Try putting a jumper on GPIO4/Ground (Pin 07-09) on the RPi. It will tell the system to mount it read-write. That way you can save settings. Let me know if that works for you.

strabbit, 2018-03-02:
Thanks -- that allowed me to save the setting. Android Auto still gives me a message about bluetooth not being paired. Are there special steps I need to pair bluetooth? I followed the [command line instructions here](https://www.cnet.com/g00/how-to/how-to-setup-bluetooth-on-a-raspberry-pi-3/), (after having the jumper in place) and it's still not working. It's not clear if there are more instructions on the settings screen because the text gets cut off.

[image]

[image]

htruong, 2018-03-02:
Oh yeah, for now, you have to do some magic voodoo to get it paired with the stereo you have and even when it pairs, it won't provide you the phone capabilities (it's called HFP - HandsFree Profile). The RPi3 Bluetooth adapter doesn't have that profile "programmed" in the hardware. In other words, it's pretty hopeless. You'll have to live with it for now.

parkerlreed, 2018-03-02:
I thought that Bluetooth support in OpenAuto was the other way around, pairing phone to the computer running OpenAuto.

I paired my phone to the computer and made sure Ofono was running and working in Pulse. That error message then goes away and I get the audio through Pulse (I don't think OpenAuto actually manges any of the Bluetooth connection, it just sees that the phone is connected, HFP working in Pulse, and thus doesn't show the "not connected" message)

htruong, 2018-03-03:
Hmm, I thought it's the other way around. I might be wrong. Let me figure out what is going on first :)

bradsour, 2018-03-05:
Sorry for my lack of understanding. Can I just connect GPIO4 (Pin 16) to Ground (Pin 9)? Because if so... this doesn't work for me.

htruong, 2018-03-13:
Fixed as of 0.2.0!

https://www.reddit.com/r/crankshaft/comments/842r3g/breakthrough_release_crankshaft_alpha_020_and

closed 2018-03-13 by htruong