opencardev/crankshaft #13: Writing changes doesn't work

KG7WVX, 2018-03-03:
Samsung Galaxy S8+ 02/28/18 Image

When booted into the Read write mode (jumping GPIO 4 to ground) it will reboot with modified settings then immediately reboot wiping all settings back to default.

htruong, 2018-03-03:
This is quite a headache. Let me figure out something to get rid of this problem once and for all. This might need some architectural change wrt how the sd card is partitioned. 

So please bear with me. :)

htruong, 2018-03-03:
Related issue #8

htruong, 2018-03-05:
I have repurposed the GPIO 4 to be dev mode. It will enable the system to read-write and do a lot of fun stuff. It should resolve this problem. Please try image 2018-03-05 and see if that works for you.

Ref: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

Hope that it helps!

htruong, 2018-03-10:
This is fixed on image alpha 0.1.7 2018-03-10 with no further action needed from the user. Enjoy :)

closed 2018-03-10 by htruong