opencardev/crankshaft #14: A triggered shutdown script for GPIO

htruong, 2018-03-04:
Some users want a hardwired shutdown button, so let's use GPIO 3 as a trigger for shutdown.

jlukanc1, 2018-03-07:
Does the "run" pins on the board not trigger a shutdown? Could they be used as these shutdown pins instead of GPIO 3? They are the 2 unsoldered pinholes by the end of the GPIO headers under the screw.

htruong, 2018-03-10:
@jlukanc1 actually no, the RUN header will hard-reset the CPU, so it will be more akin to pressing Ctrl+Alt+Del on DOS. We can use it to turn on the Pi, but it cannot turn off the Pi (Technically, I can make it turn off the Pi with an extremely ugly hack, but I'd rather not do that).

Maju3, 2018-03-26:
Well GPIO3 wakes up the pi from halt state BY DEFAULT.Without any script.So when you implement a SHUTDOWN script for GPIO3 into crankshaft then you get boot AND shutdown with GPIO3.this makes gpio3 to an on/off button you know

htruong, 2018-03-26:
@Maju3 Very interesting. Let me figure out how to listen to the GPIO!

Maju3, 2018-03-27:
Thank You!
The best way would be to listen to GPIO3 for CHANGING STATE.
Let me explain:
You use a simple 12v relay and connect the coil of it to the IGNITION line of your car.
To the button contacts of the relay you connect pin5 and pin 6 from gpio.

Then it works like this:
You start the ignition of your car.The relay gets 12 Volt and Closes the Switch contacts.This bridges pin5 and pin 6.GPIO3 is LOW and the PI boots up.GPIO3 is low as Long as your ignition is on.

Now when you turn off the cars ignition then the relay looses 12v and opens the Switch contacts.GPIO3 is HIGH now and the PI shutsdown.

pretty simple and very cheap from Hardware side.

But i have no clue about Scripting anything so implementing to listen GPIO3 STATE CHANGEMENT is to you;)

tehniemer, 2018-03-27:
What's the risk of just using the switched 12v from the ignition to supply a 12v-5v power converter and just letting ignition kill power to the pi?

DieKatzchen, 2018-03-27:
About the same as yanking the cord out of your PC while it's running. Some data corruption, possibly to something vital to the Operating System.

tehniemer, 2018-03-27:
I guess I'll be the guinea pig, that's my plan for now since I'm hardwiring
everything behind the dash.

Maju3, 2018-03-27:
Crankshaft is a Read only System so the risk of data corruption is low.Anyway i think a clean shutdown is a better solution as just cutting the power from the pi.
And as said above you just Need one simple relay;)

stramazzo, 2018-04-01:
A cheaper and as safe and releable solution alternative to the relay is an optocoupler (4N28 or similar).
LED side to the 12v/24v with resistor, transistor side emittor to gnd, transistor side collector to GPIO3 + pullup resistor to 3v3.
Note: To put the raspberry in halt, does not stop the dc/dc converters. A very small current is still drawn (next week I'll try to make some measurament).
Anyway on my motorbyke, I think I will kill the power to the Raspberry, I prefere the small risk of data corruption to the slow battery drain.

Maju3, 2018-04-01:
I really can recomment the M3ATX psu:)
It is made for ATX Motherboards but it provides 12 Volt AND 5 Volt.
And it has setable Timings for shutdown 12 Volt and 5 Volt lines.
And the best of all it shuts completelly down itself......ZERO battery drain;)
Just Google for it at cartft.com

DieKatzchen, 2018-04-01:
If I'm understanding this right, you could put an atx power off signal as the last thing in the shutdown script, and have a delay long enough to be sure the pi is in a safe state. Is that about right?

Maju3, 2018-04-01:
Well the M3ATX has one cable which you can connect to an atx motherboard where you normally would connect the power button.if you turn on the cars ignition then the m3 closes the contacts of this cable.same again when you turn the ignition off.so it just works like pushing the power button on your home pc you know

Maju3, 2018-04-01:
But yes you are right.
The m3atx gives a power off signal on this cable and the cut power delay is long enough to be sure that the pc or the pi is shuted down

Maju3, 2018-04-01:
I am still using the m3atx on my windows based car pc with a mini itx motherboard and it works great.
Now i am going to replace it with a PI and crankshaft.
Just need anyone tell me how to create a shutdown script for the m3atx:)

i have zero knowledge about scripts.
I am just a hardware guy;)

Maju3, 2018-04-02:
BTW there are more automotive psu like the M3ATX.
Just google on "www.cartft.com".
There is the M1ATX and the M2ATX next to the M3ATX.
The difference is its size and output power in Watt.
I bought my M3ATX on ebay for about 30€.
It was used when i bought it but that was about 7 years ago and it still works without a problem.
So i can absolutelly say this is a rock solid hardware.
And remember this psu has 5 volt AND 12 volt output;)

Maju3, 2018-04-02:
I got a solution to use GPIO3 as on/off button:)
check my Video:https://youtu.be/mIdApFuj058
And be Patient with my poor english;)

To use GPIO3 (pin5/pin6) as button just edit the config.txt file in root of sd Card.
write this into config.txt:
dtoverlay=gpio-shutdown

That is all;)

knotbeerdan, 2018-04-10:
I am planning on following how this person did it. You have to watch both videos to see how he handles the start up and shutdown of the Pi. So far I have the shutdown script working, and I am planning on the octocoupler sensing my ignition wire, so when I turn the ignition off, my Pi will run off my delayed accessory circuit and shut down before I open my doors. Right now I just have a normal momentary button connected to pin 3 and ground.

Video - https://youtu.be/Puk_pzMGd7c

Additional scripts he mentions - https://gist.github.com/Everlanders (would be nice if these could be worked into the default image)

Raspberry pi auto shutdown script - https://howchoo.com/g/mwnlytk3zmm/how-to-add-a-power-button-to-your-raspberry-pi

In the end I am thinking of using optocouplers to sense the headlights to dim the screen, and if I add a backup camera, I will have an additional one detecting my reverse lights power.

htruong, 2018-04-10:
Add a line that says:

```
dtoverlay=gpio-shutdown,gpio_pin=21,active_low=1,gpio_pull=up
```

or something like that to ``boot/config.txt` and you'd be done :)

htruong, 2018-04-10:
Done: https://github.com/htruong/crankshaft/wiki/Boot-and-Shutdown-the-Pi-with-ignition-key

closed 2018-04-10 by htruong