opencardev/crankshaft #15: Access to terminal for settings change

KG7WVX, 2018-03-04:
Galaxy s8+ IMG 2018-03-02
Maybe make the terminal accessible so people can change things such as screen rotation through boot/config.txt and master volume through alsamixer.

htruong, 2018-03-04:
Yep, I know someone will soon report this bug :) For now if you want to access the terminal, please download my script [chroot-to-pi.sh](https://gist.github.com/htruong/7df502fb60268eeee5bca21ef3e436eb) and run it under Linux on your SD card. Or wait for the next release that will surely allow you to do that :)

htruong, 2018-03-05:
This is positively fixed with the dev mode jumper. 

Ref: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

closed 2018-03-05 by htruong