opencardev/crankshaft #16: Audio stuttering with new build of OpenAuto

htruong, 2018-03-04:
The pain is not over yet. When listening to podcasts and driving at the same time, audio stutters badly. Revert OpenAuto's change to resolve this problem.

htruong, 2018-03-04:
Suspecting commit: https://github.com/f1xpl/openauto/commit/88aeb591b4000fda77cb0bed145e54d4f64a9a2a

htruong, 2018-03-07:
The commit was not the problem. The problem arises from one CPU core being overloaded when drawing maps.

rcrichton, 2018-03-08:
I can confirm this happen when Google Maps is animating or during app transitions.

Any idea on how this can be improved? If a core is being maxed is the Pi3 just too under-powered or are there some optimisations to be had?

htruong, 2018-03-08:
I honestly don't know yet. 

The code on OpenAuto seems reasonable and the only thing I could think of is to move audio decoding to another thread or implement some kind of frameskipping. I do not yet understand how this pegs the CPU of the Pi while none of the tasks actually require that much processing power.

htruong, 2018-03-11:
Hey @rcrichton, I have a new build here that will skip the Pi for audio and use the phone's normal audio output. It will help with this issue. Please refer to #25 to test :)

EveAeternam, 2018-03-11:
My Google maps runs fine strangely enough. I haven't tested waze, but it seemed to work fine for me. Sometimes the volume would lower for too long when the GPS voice would talk, but it only ever happened once or twice

rcrichton, 2018-03-13:
Flashed the new image and it's been buttery smooth in my car even while trying to make it stutter. Nice work @htruong :)

icecube45, 2019-02-18:
@htruong Would you be able to point me towards your build steps for openauto? I'm facing the same maps core maxing on my fork of openauto and can't seem to track down what changes were made in the build process.