opencardev/crankshaft #17: Docker build

htruong, 2018-03-04:
It would be nice to eventually be able to have an automated build system in Docker or any continuous integration system. There are more scripting needed before that becomes a reality.

htruong, 2018-03-07:
I'm bumping this to the nearest alpha release. Having a reproducible environment means that more people will be able to assist with developing Crankshaft, thus justitifies the prioritization of the ticket.

I hope to not having to be the sole person who can produce the whole image. This will certainly help with times when I'm busy working on my personal work and life.

htruong, 2018-03-15:
Building in an emulated environment messes up some assumptions about the capabilities of the CPU, resulting in janky performance. This idea is now pivoted to building on an RPi 0 raspbian-lite continuous integration environment.

htruong, 2018-04-01:
0.2.1 provides a script to build Qt5 on raspbian. We will need to think about how we want to build openauto and aadsk because they will change in the future.

Almagnus1, 2019-03-16:
Have you considered using cmake to generate a build?

htruong, 2019-03-17:
I am working on buildroot-based builds. This is going to be somewhat in the long run.

matt2005, 2020-10-01:
Setup and working

closed 2020-10-01 by matt2005