opencardev/crankshaft #18: x-axis flipped

floyduww, 2018-03-05:
I am using an egalax hdmi screen and usb touchscreen.  Everything is working great except my x-axis is flipped for the touchscreen.   Is there any setting to flip the x-axis?

htruong, 2018-03-05:
Try to see if this helps, basically you will have to launch autoapp with `QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event5:invertx=true`. To do that, edit `/etc/systemd/system/autoapp.service`, add a `Environment=[above]` in the service section. 

The current version of Crankshaft has a known limitation that does not allow you to drop to the shell so you can login and edit the file (you can always mount it in Linux though). I plan to release another version within a week to address that limitation. If you can't find a way to edit that file, please bear with me for a couple of days to resolve that issue.

Ref: http://doc.qt.io/qt-5/embedded-linux.html

htruong, 2018-03-05:
Related issue #15

htruong, 2018-03-05:
@floyduww , try this.

First, get the 2018-03-05 image. Get a jumper or a jumper cable somewhere and put it to dev mode: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

It will drop you to the shell after a restart. You can now `ssh` into it. Try playing with it with something like this

`/usr/local/bin/autoapp -plugin Evdevtouch:/dev/input/event3:invertx`

try replacing event3 with 0->5, see if anything works.

floyduww, 2018-03-05:
/usr/local/bin/autoapp -plugin Evdevtouch:/dev/input/event0:inverty    has me closer. It does flip the axis and touches are in the right spot.  But this causes autoapp to crash after several touches. FYI it was my y axis not x axis, i had them swapped in my head.

Next I will try using tslib and also the x11 compat mode.  I know that this touchscreen does work as I have been using it with xbmc for a while.   I will do more work with my other setup to see how it is handling the touchscreen.

htruong, 2018-03-06:
Interesting. At some point I have to include gdb on this so you can help me with debugging this crash :) Let me know if X11 works though.

htruong, 2018-03-06:
Note that the current build of crankshaft does not include tslib. If you need tslib, I'd be happy to include it in the next build. I will need to disable it for everyone else because it seems like evdev and tslib does not automatically play nice with each other.

floyduww, 2018-03-06:
having tslib as part of your build and disabled would be awesome.   Saves me alot of time of re-creating your build environment.   And I could test to see if that helps.  I hope to get to being able to test X11 in the next couple of days.

htruong, 2018-03-10:
X11 works, and X11 is easy for now, so let's just all be happy with X11 :) Tslib will be available next time I get around to it.

closed 2018-03-10 by htruong

floyduww, 2018-03-11:
I switched to X11 on the 3-10 build.   But still have the y axis inverted.   Do you know what parameter makes it work for X11 ?

htruong, 2018-03-11:
Hmm, I don't know... Maybe this? It seems like the problem is with the touchscreen in the first place then...

https://www.raspberrypi.org/forums/viewtopic.php?p=425962#p425962

floyduww, 2018-03-11:
got it. using X11 mode and these changes

apt-get install xserver-xorg-input-evdev

cd  /usr/share/X11/xorg.conf.d/
mv 10-evdev.conf 45-evdev.conf

vi 99-configuration
`Section "InputClass"
  Identifier "calibration"
  MatchProduct "eGalax Inc. USB TouchController"
  Option "InvertX" "0"
  Option "InvertY" "1"
EndSection`

htruong, 2018-03-12:
:)

AtlanticZeiser, 2023-04-24:
Hello, 

 As far as I am not very good with coding, I went via the simpliest way... The resistive touchscreens are usually 4 pin, and conasequently they are:

1. x+
2. y+
3. x-
4. y-

I had the same problem but was not smart enough to insert the software fix, wo I've just changed the y+ and y- wires and voila.

Probable it will be helpfull for somebody like me ) Have a nice day!