opencardev/crankshaft #23: Will it work , when connected to a LCD monitor  ?

prabhuvd, 2018-03-06:
-  The system doesnt startup and stops with a  error screen.
-  Did not get a chance to connect to the phone yet Galaxy Edge S7?

tehniemer, 2018-03-06:
I had mine working with a DELL U3415W, no touch though obviously.

contherad, 2018-03-06:
Same here, worked plugged in to a Dell 27". No touchscreen.

prabhuvd, 2018-03-07:
After reading other blogs about the error on boot up screen - i'm going to try reformating the SD card

htruong, 2018-03-07:
The rfkill stuff is fine, just ignore it. What happened when you plug the phone in when it gets to the android screen?

htruong, 2018-03-07:
By the way - I understand you now. That's not the error screen. It just tells you to connect the phone over USB to it :)

htruong, 2018-03-08:
New screen proposal.
[image]

EveAeternam, 2018-03-08:
Love it! What about displaying a clock or something low key instead?

htruong, 2018-03-08:
The Pi actually doesn't have a hardware clock. It doesn't know by itself what time it is now. I'm quite irritated by it actually -- it means we have to "tell" the Pi the time from the smartphone, that's the only way.

farmaan999, 2018-03-08:
You should get rid of the droid at the bottom corner and just add the new Android logo on the centre of the phone. To give it a modern look. Also the graphic of the usb looks unproportional.

htruong, 2018-03-08:
I will fix the non-proportional USB plug - do you mean the top of it is long?

I am afraid I can't use the android / auto logo. This is an independent project. We really, really want Google to be cool with us. Using the droid I think is pushing it. That's why I tried to put it on the side, so it's a derivative work.

benmprojects, 2018-03-08:
Looks Great.. Could we have a basic music player and settings menu displayed here like a real AA headunit would be like before the phone is plugged in.

htruong, 2018-03-08:
@benmprojects If you want a standalone music player, perhaps this can entertain you: https://github.com/viktorgino/headunit-desktop :) It's free software too so I'd encourage you to give it a try and give the developer who worked hard on this some feedback. I don't plan to implement that feature in the near future because it will increase the complexity of the software significantly, yet provide features that the phone already provided.

I haven't seen the settings screen, but that'd be something nice, I agree. I will look into it and see what I can do.

EveAeternam, 2018-03-10:
Maybe make the GUI buttons on the side look friendlier?

htruong, 2018-03-10:
@EveAeternam what's your suggestion?

EveAeternam, 2018-03-10:
I was thinking some graphics that look less like a DIY project and more like something you'd actually find in a car. For example, maybe have the buttons hide after 30 seconds of inactivity and have them popup after pressing any part of the screen. Also how about eliminating some of the boot console? I'm pretty sure there's a way to bring it to minimum info, I don't know if unclutter too would help.

htruong, 2018-03-10:
@EveAeternam Haha! It looks like a DIY project because it is! 

I'm just joking, but yeah, points taken. I will try to polish the interface soon with the hiding of buttons as you suggested. I want to have the screen automatically sleep at some point too, but I haven't found a great way to get it done.

Having the splash screen coming up earlier is actually much harder than I've ever imagined... So that's that. I'm sure there are ways to get rid of the boot console, but I think that might involve repacking of the `initrd`, which is going to be somewhat painful. I will keep that in mind. So little time, so much to do :)

EveAeternam, 2018-03-10:
@htruong Isn't achieving a non-DIY look the ultimate compliment for a DIY project though? :)

How about this: instead of powering the official LCD via a 5v PWR pin, why not have it powered via a GPIO pin and have the pi turn it on only after 10 seconds (for instance). Also, and this might be a big one: on screen volume buttons! For some reason my pi's volume is vero low by default, and if I crank the car up, it saturates my speakers faster.

As for the the screen sleep, I'm assuming you're using LXDE. Why not have xscreensaver or whatever it's called intervene after 60 seconds when Android Auto isn't on? The actual screen can be put on a power cycle too, but I hardly see the point since the car's alternator should keep enough power to the on-board Acc's.

Another quick suggestion: would it be possible to have a GPIO pin dedicated as a safe shutdown/boot switch? This way one could wire the car's ignition wire and the constant power to allow the pi to boot up and shutdown safely once the car turns on/off.

EveAeternam, 2018-03-10:
As for the Options menu, there's a bunch of things I can think of. One of which, what about - instead of jumping pins which requires pulling the unit out and/or installing a button, have the dev mode boot-up by holding the power button?

Out of curiosity, what are you using to code the main GUI? I'm genuinely impressed, kudos on the great job! You have a fan :)

EveAeternam, 2018-03-10:
[image]

What about something like this? These are obviously prototyping icons, but I can work on making some custom ones from scratch if you'd like :)

htruong, 2018-03-10:
I'm uploading alpha0.1.7 now :)

You'll probably like this: https://github.com/htruong/crankshaft/wiki/Customizing-Crankshaft

Here is mine: [image]

EveAeternam, 2018-03-10:
I love it ❤️  this is exactly what I had mentioned in the subreddit 🤣 
[](https://www.reddit.com/r/crankshaft/comments/83cnyf/improve_the_options_menu/)

htruong, 2018-03-10:
As this issue is resolved as of alpha0.1.7 image, I will close the ticket now. 

@EveAeternam The power on trick is mentioned [on the tricks](https://github.com/htruong/crankshaft/wiki/Hidden-tips,-tricks,-settings,-etc.#turn-on-the-pi-with-a-button) page. Power off by a button is not implemented yet, but is on the radar. I know you have more questions in here, but let's move it to reddit. I might also ask you to help with designing the UI :) Thanks!

closed 2018-03-10 by htruong

rubenverhoef, 2018-04-30:
> The Pi actually doesn't have a hardware clock. It doesn't know by itself what time it is now. I'm quite irritated by it actually -- it means we have to "tell" the Pi the time from the smartphone, that's the only way.

@htruong i would really like a clock in the home screen, i have ordered a DS3231 RTC for the RPI which should know the time (after an one time sync). With this cheap component we can make it possible.