opencardev/crankshaft #25: Music volume lowering for navigation

floyduww, 2018-03-06:
The volume lowering of music from Google Music is delayed by a few seconds from when navigation (Google Maps) info is played. 

Example:
Turn left in 400 ft.   And then a couple seconds later the music volume will decrease.

This happens when on the main screen, when I have the map displayed I get stuttering like issue #16 .   I have not yet tried to see if the issue happens with other notifications like text messages.

I did notice a long buffering with music also.  I press skip to next song and it would take about 3-4 seconds to change what is coming out, however the display showed the next song.  This could be a related buffering issue or a completely different issue.

htruong, 2018-03-06:
Yep, I think it buffers a little bit too much. It's very annoying, I'm pushing/testing a change that will lower the amount of buffering needed.

floyduww, 2018-03-06:
I just noticed no buffering issues and instant song skip in google music when no other tasks using audio output.

htruong, 2018-03-07:
I wonder if you can help by putting the device into dev mode, and edit `/etc/pulse/daemon.conf` accordingly, beginning with [this](https://github.com/htruong/crankshaft/blob/master/precompiled/pulseaudio_daemon.conf) as a starting point, to see if anything helps at all. `fragments*fragments-msec = total buffering`. The latest image has 10*10. Now I'm thinking of doing 8*5. There might be a better setting. But I haven't gotten much time to test. I wonder if you could help testing.

floyduww, 2018-03-07:
A test of the fragments changes did not help.  The volume lowering and a song skip are still delayed when using navigation.

htruong, 2018-03-08:
I tested and it didn't help either. I will need to dig into OpenAuto's code to figure out what is going on here.

floyduww, 2018-03-09:
I think I have narrowed down the issue a little more to delayed audio buffers.  The audio ducking of the music is done on the phone and sent as part of the audio stream.  So if the music gets delayed at all, like when starting maps or other big draw tasks, then the playing of the stream falls behind.

htruong, 2018-03-10:
@floyduww Try this workaround for now: 

I have pushed an experimental build, in which OpenAuto will not take over the phone's Media and Speech output (so the phone continues outputting by whatever means it normally does, bluetooth, headphone jack, etc).

Download at: https://github.com/htruong/openauto/releases/tag/v0.0.1-aud-exp

floyduww, 2018-03-11:
The does make the audio work, but it does not pass audio out via usb to the Pi.  So you can only use the other methods for audio out.

htruong, 2018-03-11:
Yep... In the long run, I still have to figure out why the CPU gets hogged.

floyduww, 2018-03-13:
I have done a full build based on the openauto directions, and I have no issues with audio ducking timing, and no choppy audio during map redraws.

htruong, 2018-03-13:
@floyduww  Yeah, I am aware that there is something unusual with the build process I have, but haven't gotten time to figure out why.