opencardev/crankshaft #27: Using Adafruit 5"/7" touch screen

smartroad, 2018-03-07:
Hi all, using RPI3 with an Adafruit 5" screen. I can get the screen working (after editing the config.txt file with the Adafuit changes), but the touchscreen - unsurprisingly - doesn't. The touchscreen registers as a mouse and although Crankshaft does register the touch it doesn't take note of the position.

Doubt this is something anyone is worried about, I just happened to have a 5" screen already 😁 Would be nice to try in car properly before maybe getting new screen 😄

htruong, 2018-03-07:
I wonder if you tried the X11 in dev mode trick? Does the screen still refuse to work then?

https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode#run-openauto-under-x11-compatibility-mode

smartroad, 2018-03-07:
I'll give that a go when I get home, away for work at the moment :) forgot to bring a keyboard with me lol brought everything else but that...

smartroad, 2018-03-08:
Just tried and switching to X11 mode works 😁 Setup the system to auto boot into X11 and all is well now, seems to be working great given it is a resistive touchscreen.

One thing I did find, some of the instructions needed sudo in front 😄

htruong, 2018-03-09:
Glad that it works for you. I will make it easy to switch to X11 next time so you don't have to go through the pain of getting into dev mode.

closed 2018-03-09 by htruong