opencardev/crankshaft #28: Unable get it work

goodboy2462, 2018-03-08:
I got a raspberry pi with no display, so thought of hooking it up to my monitor with HDMI cable. I could boot it up and connect my pixel to the device. I get the message that Android auto is enabled, on my pixel, but the display  on the monitor still shows nothing conncected. I have a mouse connected too, but it won't do much most of the time.  

- What phone do you have? What OS version? Google pixel; Andrid 8.1

- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode
- Please provide any further information that you might find helpful.
[image]

goodboy2462, 2018-03-09:
Tried with another phone, LG G6, Android 7.0 and i am seeing the same thing. 
It shows "Connected to your car" on the notification. But not showing anything on the display

htruong, 2018-03-09:
Are you comfortable with the command line? There are some simple tricks you can try here to make it work in X11: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

Try to see if that helps it. If it does, I'm willing to make it easier for you in the next release :)

htruong, 2018-03-10:
OK guys, the new image is out. Now you don't have to worry about dev mode. It should "just works."

I wonder if you guys help me test whether it works by downloading the new 0.1.7. image and putting a jumper on put a jumper on GPIO26-Ground (they should sit next to each other, last two pins of the same row). This should tell Crankshaft to start OpenAuto under X11, and let me know if that helps with your problem. :)

goodboy2462, 2018-03-10:
Excited.. going to try this

goodboy2462, 2018-03-12:
Hmm no luck. Tried with 3 phones so far. was waiting to buy a touch screen for Raspberry pi, if this works. 
[image]

goodboy2462, 2018-03-14:
My assumption is, the display won't matter and it should automatically list the supported apps when i plug in my phone. Correct me if i am wrong

My phones are getting the notification that it got connected to a car though and Car Mode is enabled in the phones.

htruong, 2018-03-14:
Hmm, I think there might be something about the screen your Pi don't like. Are you sure you're in the X11 mode? When it gets to the "plug phone in" screen, press space bar once. If you see a black mouse cursor, you're in X11. That way it's more forgiving on screens.

goodboy2462, 2018-03-14:
Yes when i press the space bar after connecting a keyboard, i can see the Cursor.. I dont want you to block your awesome progress you made, if this is just me. :)

goodboy2462, 2018-03-22:
I got my touch screen and it started working with the touch screen!!!
This is awesome.

closed 2018-03-22 by goodboy2462