opencardev/crankshaft #29: Returning to Android Auto

EveAeternam, 2018-03-10:
Currently very little use for this, but perhaps a button on the main screen to return to Android Auto if a phone is connected? Since Android Auto allows us to return to the OS using a button, it seems only fitting to allow the opposite too!

htruong, 2018-03-11:
I think this is one thing that would better be filed with f1xpl's OpenAuto :) But let me see what I can do.

DJFliX, 2019-01-19:
@EveAeternam In the current builds of Crankshaft-ng your USB-device will show up as a tile and you can press it to reconnect to Android Auto. Is this the functionality you described? If this is the case please feel free to close this issue. Otherwise, please elaborate on what you are missing in the current implementation.

closed 2020-10-01 by matt2005