opencardev/crankshaft #30: Custom Device/Key Mapping

EveAeternam, 2018-03-10:
I don't see this as currently implemented (*correct me if I'm wrong!*) but would it be possible to implement custom key binding? Like if you have a USB button connected to the Pi, or a USB gadget, having said gadget do an action like "Pause/Play" or "Mute" and volume controls would be wonderful. That way people can integrate their own buttons in the car!

farmaan999, 2018-03-10:
Also if that button could turn on the listening feature for Google Assistant

EveAeternam, 2018-03-10:
As long as it's mappable and bindable, I guess anything is possible!

htruong, 2018-03-11:
You already can. Just that it will require some tinkering.

https://www.sparkfun.com/tutorials/337

Then map it to OpenAuto's actions. I might get some GPIO stuff going later and it's in the plan too. I plan to resolve #14 at some point.

Coedy, 2018-03-14:
I'm sure everyone has seen this already but he does talk about shutdown in his newest video (he uses openauto).
https://youtu.be/jFXxtxoi_pc

shimawn, 2018-03-15:
What about connecting a bluetooth keyboard using something like this: 

https://www.adafruit.com/product/1535

This could make it possible to build some controls into a steering wheel or anywhere else that is impractical to extend a USB cable to.

vistazifta, 2018-03-21:
I'm hoping we can do this via gpio so that I can use my steering wheel controls and replace my oem headunit altogether