opencardev/crankshaft #33: USB sound card now distorted on latest version (built in fine)

a-haan, 2018-03-10:
- What phone do you have? What OS version? Pixel XL 2
- Did you try to enable and run autoapp under X11 dev mode?  no
- Please provide any further information that you might find helpful:

Previous build had no sound problems using the USB sound card, I suspect the sound amplification on this build was for all not just the built in Aux

a-haan, 2018-03-10:
sound card is https://www.amazon.co.uk/gp/product/B01N905VOY/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

htruong, 2018-03-10:
If you're familiar with the command line, can you help me edit a few files to see a "sweet spot?"

byransays, 2018-03-11:
Same sound card. Fire away with some sweet spot ideas.

a-haan, 2018-03-11:
the sweet spot for me was the previous version, just reverted back and it's perfect again...

htruong, 2018-03-13:
New version is less jealous about the volume level. Please help me giving it a try :) 

https://www.reddit.com/r/crankshaft/comments/842r3g/breakthrough_release_crankshaft_alpha_020_and

Mearman, 2018-03-13:
@htruong does the new release fix the volume and interference issues?

Mearman, 2018-03-13:
> It should.

I'll give a go. Thanks

a-haan, 2018-04-09:
still distorted on the latest build, can you tell me what file I need to edit to adjust the amplification?

htruong, 2018-04-09:
@a-haan can you try editing `/etc/pulse/default.pa` and remove the line `set-sink-volume 0 52428` or tune it to a smaller number?

a-haan, 2018-04-09:
Thank you for your help, found a sweet spot at 32428 - music is now loud and crystal clear!

poxin13, 2018-04-22:
Also had this issue using a Hifiberry. Ended up commenting out the set-sink-volume line and setting volume in alsamixer. Also changed pulseaudio to high-priority & realtime with ffmpeg as the remix per openauto suggestion.

closed 2021-11-27 by github-actions[bot]