opencardev/crankshaft #34: Request to have colour preset's or saturated mode

a-haan, 2018-03-11:
I'm used to OLED screens, the 7" official touchscreen is very washed out colour wise, having the option to make it more vibrant would be ace!

htruong, 2018-03-12:
Well, this is not an OLED screen -- But how about this?

https://www.amazon.com/SmallHD-MON-702-OLED-inch-Display-Monitor/dp/B06XJ9D5D8

;-)

parkerlreed, 2018-03-12:
I think the point @htruong is trying to make is that there's not much you can do to colors to make a dull screen better. (Plus does the Pi even have control over that?)

htruong, 2018-03-13:
Well, with regrets, this will be the first issue I won't fix -- I just can't see any way I can fix this. Please let me know if you have better ideas how to get around this. :(

closed 2018-03-13 by htruong