opencardev/crankshaft #35: Feature: Remove all messages during boot, for a more factory look

rcrichton, 2018-03-13:
It would be great if we could display a splash screen or at least no messages during boot. This would make the bootup look better for use day to day. Of course in dev mode those messages may be useful however, outside of dev mode they shouldn't be necessary.

Perhaps something like this would help: https://raspberrypi.stackexchange.com/questions/59310/remove-boot-messages-all-text-in-jessie/59311

EveAeternam, 2018-03-15:
Quick suggestion: what about modifying the splash script? It could display the crankshaft splash image, then if the user wants, an optional one that could match their car model or etc.

jlukanc1, 2018-03-16:
This could be done similar to how Raspbian Pixel has the splash screen, just add the ability to change it and maybe add a few premade splash screens.

htruong, 2018-03-16:
That'd be possible. There is already a pull request in, I will review it and merge. This weekend will be the development/testing/refine weekend :)

htruong, 2018-04-01:
Done!

closed 2018-04-01 by htruong

Richard6360, 2019-01-29:
How is this archieved ?? I stil have boot text messages in cs-ng alpha

hawkeyexp, 2019-01-29:
usually you will only see the splash -. only if something  is triggered like dev mode, sound device missing etc. the splash will be hidden to show the notify.

Richard6360, 2019-01-29:
I did a new instal.Al I see is okNo warnings I believe.Is there an option to show the splash in stead of message?

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/opencardev/crankshaft/issues/35#issuecomment-458651649",
"url": "https://github.com/opencardev/crankshaft/issues/35#issuecomment-458651649",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

hawkeyexp, 2019-01-31:
Add a video and debuglog please...