opencardev/crankshaft #37: Android Auto does not always start on latest release

pumpkintechlabs, 2018-03-14:
- What phone do you have? What OS version?: Pixel XL 8.1
- Did you try to enable and run autoapp under X11 dev mode?  no
- Please provide any further information that you might find helpful.

Latest release 0.2.0 - plugging in the USB into the phone does not always trigger Android Auto to start. Restarting Pi sometimes help, sometimes doesn't.

P.S. such an awesome project, thanks to everyone for their hard work on this!

htruong, 2018-03-14:
This seems to be a problem to more people than I expected. This will be investigated and resolved in the soonest time possible.

devangapatel, 2018-03-16:
Also, this may or may not be technically possible.... I have noticed that if the phone is plugged in before I turn on my car engine (which starts crankshaft), that Crankshaft is not recognizing that the phone is already plugged in. I have to unplug the phone and replug it for Crankshaft to launch Android Auto functionality

with the 0.2 build, if the phone is plugged into the USB slot

parkerlreed, 2018-03-16:
@devangapatel If I'm reading this correctly, it's being worked on. 

https://github.com/f1xpl/aasdk/issues/13

https://github.com/f1xpl/openauto/issues/41

htruong, 2018-04-01:
@pumpkintechlabs Please help me test the new version 0.2.1 :) I haven't encountered any time where OpenAuto doesn't detect the phone. So let me know if that works for you too!

stramazzo, 2018-04-01:
@htruong 
-Nexus 5X
-Crankshaft 0.2.1
It allways detect the phone. No issue noted yet.
Amazing work.

matt2005, 2020-10-17:
This is resolved in alpha 55

closed 2020-10-17 by matt2005