opencardev/crankshaft #38: Wavshare 5 Inch LCD Works, Touchscreen Does Not Work

KE4NYV, 2018-03-15:
- Nexus 5x Android 8.1.0
- X11 Mode did not work with jumper.  In fact it hung up when I tried to plug in my phone.
- Using these LCD settings:

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=1

GPIO drivers are not working (might not be included in the latest build - Openauto 0.2.0 2018-03-13)

Even though touch does not work, it appears to be pulling data from my phone.  Displays my local weather information after Android Auto starts.  I can't get past that screen since touch is not working.

Is there a way to SSH in?  When I attempt a connect, connection is refused.

htruong, 2018-03-15:
Certainly, please enable dev mode to SSH in and figure out what is going on. https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

Perhaps you want to install  `xserver-xorg-input-evdev`. If it works for you, I will be happy to include in the next release.

KE4NYV, 2018-03-15:
I got it working using the Waveshare driver.

1. Download:  http://www.waveshare.com/w/upload/0/00/LCD-show-170703.tar.gz
2. Copy that tarball to the root of the microSD (I used my Windows box and a card reader).
4. 5 Inch LCD header covers pins.  I soldered on a 2 pin header to back of RPi to accept a block jumper.
3. Enabled dev mode by placing jumper on 2 pin header and boot up Pi.
4. Use PuTTY or other terminal to SSH into Raspberry Pi.  Login with: pi/raspberry
5. Follow these steps to install driver:

tar xzvf /boot/LCD-show-*.tar.gz  
cd LCD-show/
chmod +x LCD5-show
./LCD5-show

6. Pi automatically reboots.  Wait for boot to complete (will go back into dev mode).  Power down, remove jumper and boot back up.  Should auto start into Crankshaft and touch should work.

That worked for me.

htruong, 2018-03-15:
Great, thank you for leaving the detailed response on how you did it so we will be able to help new users. I will take note of this and will link to it once I figure out where to put all the tutorials in.

- KE0BVC

closed 2018-03-15 by htruong