opencardev/crankshaft #39: Day/Night Phone-Controlled is ignored - have to manually set for dark/light mode

a-haan, 2018-03-15:
- What phone do you have? What OS version? Pixel 2 XL, 8.1
- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode- no
- Please provide any further information that you might find helpful.
none

measle, 2018-03-15:
Same Problem: LG G4 H815 Running 7.0 (H-ROM V30 PORT V2.2)
Came across this but have not tried it:
https://play.google.com/store/apps/details?id=gb.aahelper.hr

smartroad, 2018-03-17:
just tried the app on my test bed unit and it seems to work ok

ahoneybun, 2018-04-02:
Is this controller with the phone? If it is time based it would make sense if the phone controls it. This seems to happen on the 04-01 image as well.

htruong, 2018-04-26:
Please follow directions on https://www.reddit.com/r/crankshaft/comments/8eze6m/heres_an_app_that_forces_android_auto_to_switch/ to have the mode changed automatically. Huge props to Emil!

closed 2018-04-26 by htruong