opencardev/crankshaft #40: Hifiberry Support

LoganFanGer, 2018-03-15:
Thanks for your good job ;-)
Will the "Hifiberry" sound card work? Can two sound cards, one for the microphone and one for the sound (Hifiberry), be put into operation?

Vitalic66, 2018-03-15:
A very good question witch i will follow. I also have a hifiberry dac on my pi and an usb soundcard for mic. I gave up on autoapp because i wasn't able to fix cracks in sound from speech and maps. But I can confirm, it is no problem to have 2 cards. You can tell pulseaudio where mic input comes from and also tell alsa which card is default output. I wasn't able to fix the problems on my hifiberry with any settings. But as far as i read you ise A2DP now for sound. Did it fix audio problems? I guess that the problems i faced came from the different stream types ( speech/maps 16bit mono and music 48bit stereo) so mixed stream produced cracks. Without speech or maps all was fine. Are speech and maps sounds still getting mixed on A2DP?

poxin13, 2018-04-22:
I'm using a Hifiberry DAC+ and a USB Microphone with no issues. I had to adjust pulseaudio configuration on the latest build to get rid of cracks in auto playback.

LoganFanGer, 2018-04-23:
Hey Poxin13, thank you for your information. Unfortunately I do not have enough insight into the topic "pulseaudio". Is there somewhere a guide that I can use as a guide?

poxin13, 2018-04-23:
Is sound not working at all for you with the DAC? Assuming you followed the instructions on the Wiki already: https://github.com/htruong/crankshaft/wiki/Hardware-support ? Direct instructions: http://pi.bek.no/HiFiBerry/

LoganFanGer, 2018-04-30:
I have now experimented and tested, have activated the Dev mode. I do not know how to proceed. Is there a workaround to get the Hifiberry up and running? Help would be great!

LoganFanGer, 2018-05-01:
Yeah....I did it. The sound is now output via the Hifiberry sound card. Really great! Of course it was quite simple ;-) I put the MicroSd card into the USB reader and with the help of the program Notepad ++ I edited the file config.txt in the main directory of the Crankshaft image. At the end of the file I added the following lines: dtoverlay = hifiberry-dacplus
In addition, I commented out the entry: dtparam = audio = on with a prefixed # in the first place.

In addition, I've copied the asound.conf file from my "normal" Raspbian image to the Overlays directory. Alternatively, this can of course also be created there. This is my corresponding file content:

pcm.! default {
 type hw card 1
}
ctl! default {
 type hw card 1
}

Note that instead of the 1 behind the two entries (... hw card) can also be a 0. You have to try that. With me it works like that.

Now I try to include the mic  ;-) Thanks for help!

Translated with Google translator, I hope you understand that. Have a nice day!

tehniemer, 2018-06-02:
These instructions worked for me, I think this issue can be closed.

https://www.hifiberry.com/build/documentation/configuring-linux-3-18-x/

hawkeyexp, 2018-06-27:
With upcoming builds based on crankshaft-ng branch system gets a management tool which allows easy setup of hifiberry boards in a way like `crankshaft audio hifiberry dacplus` :-)

hawkeyexp, 2018-07-09:
Released with CS-NG

closed 2018-07-09 by hawkeyexp