opencardev/crankshaft #42: Feature Request - Custom Application Links/Buttons on the Left

nickorooster, 2018-03-15:
While I have an Android phone that's compatible, I have friends and others that don't, and sometimes I might need my phone not to be plugged in (for one reason or another).

It might be useful to allow buttons to be placed on the main screen that link/point to other raspberry Pi apps so that, say, a different music player can be launched, or custom scripts can be run, like OBD tools etc.

devangapatel, 2018-03-16:
This would be a great add-on to the functionality of the Pi.

htruong, 2018-03-16:
I agree. Let me see what's the best way to implement this feature into OpenAuto so that it works for both upstreams and for Crankshaft.

chrisfromwa, 2018-06-11:
I would love to see something like this

Not sure if there is any type of app out there,  or hardware for that matter. 

But I would love to see where,  we could exit out of CrankShaft and be able to start an app that allows you to tune AM/FM.  Basically just like an OEM headunit in cars now when you're not using Android Auto.

j-ak-e, 2018-06-15:
@htruong I'd say this is outside the scope of Crankshaft, and it _is_ supported by other projects, such as [headunit-desktop](https://github.com/viktorgino/headunit-desktop)

Aaron407s, 2018-06-20:
Sorry if this is hijacking with a slightly different approach, but would it be possible to integrate a screen mirror or external screen mode to control and display the phone on the external touchscreen? Then android apps like torque could be run and controlled without having to be naively supported by the pi. It would still require the phone to be connected, but it would be great to be able to access it with the bottom right menu button in AA that currently just gives the "return to android auto" option. Just a thought anyway.

toufou54, 2019-05-31:
Same for me, it will be great to customize some buttons to launch Kodi, torque or other obd software, and chrome links to http for examples