opencardev/crankshaft #44: Brightness control in Crankshaft

devangapatel, 2018-03-16:
Would be nice to be able to control brightness of the Pi on the crankshaft UI without having to go into dev mode. This would help several of us from having to go thru adding a physical momentary button (like what everlander did and controlling this thru the GPIO ) 

Maybe even set a brightness profile based on time of the day?

htruong, 2018-03-16:
Manual brightness control is possible. Automatic brightness control is impossible. Refer to https://www.reddit.com/r/crankshaft/comments/84drzu/crontab_and_crankshaft/dvoqo41/

devangapatel, 2018-03-16:
Manual it is then! But we need to be able to do this without the need of GPIO or external physical switches. :-)

htruong, 2018-04-01:
Done as of 0.2.1!

closed 2018-04-01 by htruong