opencardev/crankshaft #46: missing "other" apps

smartroad, 2018-03-17:
I noticed that AA should be able to use other apps, such as Skype. However I have not been able to find Crankshaft (using 0.2.0 and a Pi3 and an Adafruit 5" LCD with touch in X11 mode). Am I missing something or are they not supported?

parkerlreed, 2018-03-18:
What do you mean by "other apps"? The Pi side of things is just streaming from the phone. It is completely up to the phone to support that.

htruong, 2018-03-20:
This suggestion might be integrated into other ideas such as #42 where the user is presented with other choices of apps. I tend to not try to overcommit on functionality because it will be very easy to be drowned in support requests. I am figuring out how to resolve this the most efficient way possible.

smartroad, 2018-03-24:
sorry for the delay in responding. This is the part section I mean:

[image]

I have got installed on my phone Skype and Kik installed on my phone and neither are showing up on this list. At least I guess this is where they should be listed anyway :)

htruong, 2018-03-24:
Oh -- I see.  Google purposefully does not allow you to use them while driving in Android Auto to keep you safe.

That part is not controllable by OpenAuto or Crankshaft. Sad to say, but there is absolutely nothing we can do about it. Unless you can convince Google to :)

closed 2018-03-24 by htruong