opencardev/crankshaft #49: Feature Request: Volume Control

danielthetrojan, 2018-03-18:
- What phone do you have? What OS version? Pixel 2XL
- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode
- Please provide any further information that you might find helpful.

Hi i am using this as primary head unit need some sort of volume control in the application

htruong, 2018-03-20:
Seems like vol control and brightness control are two must-haves for the functionality. I am investigating how to implement them the most efficient way.

htruong, 2018-03-21:
I will also implement a backlight control mechanism too. This is bugging me and seems like a big usability problem.

byransays, 2018-03-21:
I'm currently using a rotary encoder for volume control in my car and a script to control volume up and down. I'm also using an auto brightness script with a light sensor and a capacitor. I can post more info later if people are interested.

htruong, 2018-03-21:
@byransays I think that would be helpful definitely. I am thinking of a software solution too, because before we figure out what it is that everyone is able and willing to afford, we want to be as conservative as possible in terms of custom hardware. It'd be very inconvenient to the users if we change the hardware recommendation once we suggested it.

byransays, 2018-03-24:
I am attaching the scripts I am using. I cant remember the sources but I have edited them a little bit. I think the autobrightness still needs to be tweaked as it is a little too bright still when its dark out.

These are set as services in systemd. 
[autobrightness.txt](https://github.com/htruong/crankshaft/files/1843890/autobrightness.txt)
[volumecontrol.txt](https://github.com/htruong/crankshaft/files/1843891/volumecontrol.txt)

htruong, 2018-04-01:
Manual brightness control is implemented to support the official screen as of 0.2.1!

albdom, 2018-04-02:
@byransays how you get run volumecontrol.service? I get an error with command: sudo systemctl start volumecontrol.service

meekys, 2018-04-05:
https://github.com/meekys/vol_overlay might be of interest. I'm planning to use it in my project :-)

htruong, 2018-04-07:
Done as of 0.2.2! Not very friendly yet, but it works. https://www.reddit.com/r/crankshaft/comments/8agvly/crankshaft_022_whats_new/

closed 2018-04-07 by htruong