opencardev/crankshaft #50: Still hear audio crack especially when lounder music is played

chaoman, 2018-03-19:
- What phone do you have? What OS version? HTC 10 (2016) running Android 8.0.0
- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode. Yes. I tried to adjust output volume using alsamixer but still without a luck
- Please provide any further information that you might find helpful.

I am using an external USB sound card Sound Blaster Play! 2. I could hear cracks when I play music through Google Play Music. The cracks and pops in the beginning of audio output are very noticeable.

htruong, 2018-03-20:
@chaoman for the time being, manually editing `/etc/pulse/default.pa` and remove the line `set-sink-volume 0 52428` will solve your problems. I am figuring out ways for Crankshaft to be more intelligent about custom hardware detection.

chaoman, 2018-03-20:
Thank you for answering. I just tried to comment out `#set-sink-volume 0 52428` in `/etc/pulse/default.pa`, but I could still hear the crackle. Do you have other suggestions?

htruong, 2018-03-20:
Oh, so that wasn't due to the volume level... First, try editing `/etc/pulse/daemon.conf`, you should see:

```
default-fragments = 8
default-fragment-size-msec = 5
```

Try increasing both to 10, restart, see if that helps. If it still doesn't, try editing `/etc/pulse/default.pa` and removing `tsched=0` from it. It's due to the customization done [here](https://github.com/htruong/crankshaft/blob/master/scripts/customize-image-pi.sh#L59) on Crankshaft.

htruong, 2018-03-21:
Great to hear that it works for you. Audio in linux is such a pain :)

closed 2018-03-21 by htruong