opencardev/crankshaft #51: Wireless Android Auto support?

DGxInfinitY, 2018-03-21:
I was wondering what the ETA on Wireless Android Auto was and if anyone has had a chance to implement it? It would be so nice honestly as I don't really wish to plug in my phone every time I get in the car. Here is the hardware I have as of right now.
- Pixel 2 XL (Android Oreo 8.1.0, March Security Patch)
- Standard Official Raspberry PI Display

parkerlreed, 2018-03-21:
It is being worked on in OpenAuto/aasdlk (the components Crankshaft uses)

htruong, 2018-03-26:
This functionality is not mainlined (brought to the official branch) of OpenAuto yet. It is still very hard to use, so even it's working now (with glitchy audio) I feel like this is not yet the prime time to integrate into Crankshaft. I will find a way to latch it into the dev mode of Crankshaft, but I don't yet have a plan to release the feature widely.

DGxInfinitY, 2018-03-27:
Sure. I get that. I'd love to test the functionality on dev mode for you and provide any aid as needed. I'll be watching for the commit.

mbt28, 2018-04-04:
I believe this function can work better in pi 3 b+ as it has 5ghz faster wifi. Audio needs latency free transfer, with 2.4ghz version I dont think it is possible to get good results.

DGxInfinitY, 2018-04-04:
@mbt28 2.4ghz is certainly fast enough for audio transfer. The only limitation of other raspberry pi's are the USB controllers. The Pi2 should be ok but other Raspberry Pi's might experience problems using USB to WiFi adapters because the USB bus controller is so slow.

partach, 2018-05-13:
Second this enhancement request. The USB-C plug on my HTC U11 is not that solid. It happens frequently that a disconnect happens. This requires fiddling with the plug and paused audio. Connecting via Wifi would really be a blessing... 
And I also will happily test if needed (have a Pi3B+ with offcial 7" touchscreen)

DJFliX, 2019-01-19:
@DGxInfinitY As of Alpha-1 the WiFi-functionality of Android Auto works with Crankshaft. Have you been able to play with it already? I would appreciate it if you could give mee feedback on whether it's working for you. If it is, feel free to close the issue. If it doesn't please let us know and/or share some debug logs so we can investigate.

hawkeyexp, 2019-01-20:
Have a look to the wiki - the wireless part is updated and explains both use cases

rppgen, 2019-03-30:
Anyone tried Android auto wireless on rasberry?

DJFliX, 2019-03-31:
> Anyone tried Android auto wireless on rasberry?

Yes. Please read [this](https://github.com/opencardev/crankshaft/wiki/Android-Auto-Wifi).

rppgen, 2019-03-31:
Thanks, is there a video out there too?
Will help a lot.

toufou54, 2019-05-31:
Work fine for me too, check the wiki (why this issue is not closed ?)

parkerlreed, 2019-05-31:
Because it still requires dev options. The intent would be to have the actual wireless pairing mode be available.

alextytan, 2019-06-18:
works fine for me too. i have a oneplus3t. i have only added wifi_country, my mobile's hotspot ssid and pass. and i have started head unit server on mobile's android auto

matt2005, 2020-10-17:
Resolved since alpha1

closed 2020-10-17 by matt2005