opencardev/crankshaft #54: Feature Request: mouse pointer in AA interface

edinn, 2018-03-22:
Is it possble to enable mouse pointer in AA interface, when the phone in connected, to navigate the UI? I would like to connect RPi to car head display via HDMI to RCA cable, which does not have touch support, and to navigate the UI using wireless mouse. I have mouse pointer before I connect the phone, but after that, I suppose only touch gestures are possible.
Thanks.

htruong, 2018-03-25:
That is a mystery that I have no idea why it happens and how can I make it go away. As strange as it sounds, I have tried quite hard to make the cursor appear, and just couldn't figure out what is wrong with it.

ansher, 2018-04-10:
Thank you for opening the bug. Same issue here where existing car display doe not have touch screen. I have read some other sites where similar issues were raised (not for AA) and related to X11 configuration issue or bug.

It would be helpful for developer to explain root cause of the issue, whether it is a bug or incorrect configuration. This will allow people to help out and focus. Thanks.

j-ak-e, 2018-06-15:
@htruong OpenAuto doesn't appear to be drawing to the actual framebuffer, if you VNC in to a machine running it, you get the menu interface with the cursor, even if the HDMI port is displaying the AA interface.

I'm guessing OpenAuto is either drawing to too high a layer, or doing some magic stuff to reduce latency

daracerz, 2018-08-21:
Interesting results after enabling X11 instead of EGL.  With intentionally having my screen dimensions wrong, Crankshaft at 480 and X11 at 1080 resolution, the mouse pointer is visible and working on the X11 portion of the screen.  But as soon as the mouse pointer enter the Crankshaft screen area, it disappears.  Possibly application specific error?  Related to OpenAuto?

Kingi-s, 2018-09-08:
for me, mouse doesnt work at all, i got an samsung galaxy a3 2016 and raspberry pi 3 b+, i wanna test everything before purchasing a touchscreen.

Hein-NonesensE, 2018-12-17:
Please make this happen. I too wanted to use crankshaft via RCA on my car's screen and planned to mount a mini trackball into the ashtray.

OpenAuto itself can be used with a cursor:
https://github.com/f1xpl/openauto/wiki/Enable-visibility-of-mouse-pointer-for-Raspberry-PI-3
But I can't get to the "toggle mouse cursor" setting. And maybe this solution needs X11.

x13-me, 2018-12-17:
As I've [https://github.com/opencardev/crankshaft/issues/54#issuecomment-397627534](previously stated), OpenAuto draws directly to the framebuffer, i.e. a higher layer than the mouse cursor, the only way I can see to "rectify" this, is to force OpenAuto to render to a lower layer, as those instructions detail.

If the instructions do not work for you, that's likely a specific issue with your hardware, and will be very difficult to troubleshoot.

Android Auto is not designed to be used with a mouse, and OpenAuto supporting one is not a guaranteed feature.

Hein-NonesensE, 2018-12-18:
Thanks for the clarification. After some more testing, my cursor even went away in the crankshaft GUI, alongside with the dreadful "not enough voltage"-symbol lighting up. Maybe in the end the solution is simple. I'll report back after fixing my power issues.

Anyway, in case this doesn't help, I found two traces to possible solutions (probably too complicated for me). One is found in a different software, that uses EGL, too:
https://blog.kivy.org/2014/01/kivy-mouse-cursor-on-rapsberry-pi/
https://github.com/kivy/kivy/blob/master/kivy/modules/cursor.py
And the other one is a hardware cursor which would be a great thing anyway:
https://www.raspberrypi.org/forums/viewtopic.php?t=225434
https://github.com/JamesH65/xf86-video-fbturbo/tree/raspi_hwcursor_pi2

x13-me, 2018-12-18:
> Thanks for the clarification. After some more testing, my cursor even went away in the crankshaft GUI, alongside with the dreadful "not enough voltage"-symbol lighting up. Maybe in the end the [...]

Hardware cursor *might* work, I can test it when I'm back (early Jan), unfortunately I don't have any hardware with me currently.

If that works, I'll see if I can fork Crankshaft with it added

Hein-NonesensE, 2018-12-18:
Thank you, that would be really cool. 

No need to hurry, I won't be able to tinker much until mid January anyway.

Hein-NonesensE, 2018-12-19:
Update: Fixed my power issues and tested a bit more. I can get a mouse cursor only if I plug in the mouse after Crankshaft has started. And it is gone as soon as I click on the settings button or plug in my phone. And the situation is the same for X11 and with OMX Layer 0 (except that X11 mode sucks).

So all my hopes are on the hardware cursor now.

Hein-NonesensE, 2019-02-14:
Sorry for reporting back late. I had to solve connection issues with my phone's pie update first.
With Pre 16, an option to enable the cursor was introduced. It works great in the CS GUI. However, I couldn't get the cursor to show when AA is activated.

closed 2021-11-27 by github-actions[bot]