opencardev/crankshaft #61: Terminal?

unknowerror000, 2018-03-30:
Hi! 
I have a TFT-LCD touchscreen. I want to use it with android auto. For start it, I have to run a script to redirect the video input to this. How can I run terminal for run that script from a USB? (sorry for my bad english)

htruong, 2018-03-31:
Activate the dev mode: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

Then it will allow you to customize whatever you want to do as if it's a standard Raspbian distribution. Normally, if you want to run a script at the same time, you need to [write a systemd unit](https://unix.stackexchange.com/questions/15348/writing-basic-systemd-service-files).

htruong, 2018-04-26:
Since I don't hear back from you anymore, I assume that you got this solved :) I am closing this ticket, but please let me know if you have further questions.

closed 2018-04-26 by htruong