opencardev/crankshaft #62: RCA resolution error

tehniemer, 2018-03-30:
Phone: Nexus 5
Build: 0.2.0
Hardware: Pontiac G8 factory infotainment system (RCA in) with touch overlay and AR1100

There is an issue with the "connect your phone" screen where it seems the resolution and touch calibration [maybe] is off. It seems that the image is inset from the extents of the display by a fair amount. This doesn't seem to be an issue when a phone is plugged in. Furthermore, the touch registration is functional, but incorrect, before a phone is plugged in and seems to be correct when AA is up and running.

I have calibrated by installing xinput_calibrator on Raspbian and copying the calibration parameters over.

No borders and touches register correctly.
[image]

Black borders, touches don't register correctly, and I have no way of exiting this screen.
[image]

parkerlreed, 2018-03-30:
RCA has a fixed resolution on the Pi. (Same vertical res of 480 but a shorter horizontal res of 720) (Compare to 800 horizontal of the default screen)

Your best best would be to play with the overscan settings in config.txt. You can do this on the fly with https://github.com/ukscone/set_overscan (It auto saves to config.txt upon exiting)

tehniemer, 2018-03-31:
What doesn't make sense to me is that the borders only appear before a phone is connected. They go away once connected and come back when disconnected.

tehniemer, 2018-03-31:
Changing the overscan helped a little, I can now at least access the buttons on the far right, but still can not see the full screen. It seems that it is not scaling correctly, or at all, for RCA.

[image]

htruong, 2018-04-01:
I think the resolution was too small - the dialog box simply wasn't expecting that resolution. Too bad the current way the dialog is designed is resolution dependent and it is not going to change soon. However, you totally can change any setting with editing the openauto.ini file directly.

tehniemer, 2018-04-01:
Where is that .ini located?

htruong, 2018-04-01:
You need to activate the dev mode (instructions at the wiki) -- then edit the `/home/pi/.openauto_saved.ini` file. The logic to save and load the file is described [here - save at shutdown](https://github.com/htruong/crankshaft/blob/master/precompiled/opt/openauto_settings_save.sh) and [here - load at startup](https://github.com/htruong/crankshaft/blob/master/precompiled/opt/openauto.sh).

tehniemer, 2018-04-01:
OK, is there a way to modify the dialog box to better fit my resolution? I guess ultimately it's not a big deal, but I'm not even sure if I can see all of it. What is all supposed to be configurable in these settings?

htruong, 2018-04-01:
The DPI setting allows you to scale the Android Auto projection interface. Bigger values = bigger buttons/targets to hit.

tehniemer, 2018-07-06:
I'm re-opening this as a feature request for CSNG.

Can we get some sort of resolution settings for those of us not using HDMI?

closed 2025-01-12 by matt2005