opencardev/crankshaft #63: Sleep Mode

Maju3, 2018-04-01:
Hi.
Just tested the 0.2.1 release.
I use a custom HDMI Touchscreen.
Now when disconnecting the phone after 30 seconds my Monitor shuts off.
Even when I reconnect the phone or type on the Touchscreen then my Monitor still is off.
The only way to make it work again is restarrting the PI.

htruong, 2018-04-01:
Confirmed bug. Silly me was making some mistakes in assuming everyone has the official screen and not taking care of the edge case.

0.2.1-1 is out now and it should wake your custom screen up. Let me know if that works!

Maju3, 2018-04-01:
Just tested 0.2.1-1 and now sleep mode WORKS and it Wakes up my Screen after replugin the phone.
GREAT WORK man!
Just take a look at the shutdown script please.
Shutdown does not work for me.

htruong, 2018-04-01:
Great to hear :) Let's track the shutdown feature on a separate thread :)

closed 2018-04-01 by htruong