opencardev/crankshaft #64: help with tslib and egalax touch

krizangrandi, 2018-04-01:
hi, someone can help me?
i'm newbie in pi world and linux

I have the y flip problem and have to calibrate.
When try to use ts_calibrate even if i touch the cross, nothing change.

I have already set correctly the touch with X11, but i would use crankshaft in "normal" mode.

sorry for bad english

htruong, 2018-04-01:
You might need to specify the TS_DEVICE for tslib to "take" it when you do ts_calibrate:

https://github.com/raysan5/raylib/wiki/Install-and-configure-Touchscreen-Drivers-(RPi)

That's the only semi-intelligent thing I can think of why didn't it work.

tehniemer, 2018-04-01:
xinput_calibrator worked for me with a eGalax controller, try the instructions below 

Install some dependencies:
````
sudo apt-get install libx11-dev libxext-dev libxi-dev x11proto-input-dev
````
Download xinput_calibrator somewhere in the Raspberry PI's folder structure.
````
wget http://github.com/downloads/tias/xinput_calibrator/xinput_calibrator-0.7.5.tar.gz
````
Unpack it and then navigate to the unpacked folder and then install it using:
````
./configure
make
sudo make install
````
After this step you should run xinput_calibrator(from Xserver terminal console: first startx then open console and then run it).
````
xinput_calibrator
````
Follow the on screen instructions(touching some points on screen) and after calibration is complete you will receive a message like this:
````
Calibrating EVDEV driver for "eGalax Inc. USB TouchController" id=8
current calibration values (from XInput): min_x=1938, max_x=114 and min_y=1745, max_y=341

Doing dynamic recalibration:
Setting new calibration data: 121, 1917, 317, 1741

**--> Making the calibration permanent <--
copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf'**
Section "InputClass"
Identifier "calibration"
MatchProduct "eGalax Inc. USB TouchController"
Option "Calibration" "121 1917 317 1741"
Option "SwapAxes" "1"
EndSection
````

krizangrandi, 2018-04-04:
Let me try this evening or in the next days.
I'll give you a response as soon as possible.

krizangrandi, 2018-04-05:
OK, now it calibrate.
and ts_test confirm the functionality
i've another problem: when reboot or shutdown or launch autoapp i lost the calibrate how i can save the priority of the driver?

when all it work, i'll do a tutorial

tehniemer, 2018-04-05:
You need to create and save a file with your calibration parameters here

```
/etc/X11/xorg.conf.d/99-calibration.conf
```

krizangrandi, 2018-04-05:
Thanks tehniemer.
As i wrote in the first post, in X11 mode, i already have the touch working well. I would use tslib to have crankshaft in "normal mode"

tehniemer, 2018-04-05:
You'll probably need to leave it in X11 for the non-official screen, AFAIK there isn't a drawback to this.

krizangrandi, 2018-04-05:
I got it, finally work!!
but, the better solution remain X11.
More precision and easy to use for galax controller.

So, i think we can close this "issue"

bfpa40, 2018-04-05:
I can report Crankshaft now working on a Raspberry Pi 3 Model B+ interfacing with a Google Pixel XL via USB Cable. Pleased..Now onto getting the Bluetooth working..

computeridea, 2018-05-08:
hi, i'm tring it, but when i rote xinput_calibrator i get the message   "unable to connect to X server"
any advise about it?

krizangrandi, 2018-05-08:
Well, make the procedure in raspbian, and then report your result calibration on crankshaft.

computeridea, 2018-05-08:
i did it
i created the folder /etc/X11/xorg.conf
i create the file 99-calibration.conf

i wrote what i had from calibration
```
Section "InputClass"
Identifier "calibration"
MatchProduct "eGalax Inc. USB TouchController"
Option "MinX" "4665"
Option "MaxX" "61027"
Option "MinY" "58140"
Option "MaxY" "9248"
Option "SwapAxes" "1"
EndSection
```

but nothing i tried to add a few option found everywhere on the net

```
Option "SwapXY" "1"
Option "InvertX" "1"
Option "InvertY" "1"
```

but it seems no effect

i tried in both mode
START_X11=0 and START_X11=1

but still same result.. 
what i'm wrong? even in raspian i can't make it works, the touch is still rotate even there after reboot

anyway i undertsood why i couldn't lunch xinput_calibration from crankshaft, date is 2016 and ./configure , make and sudo make install throw ecception becouse of it, in raspina everything went fine.. anyway if i use  date -s with correct date raspberry instantly shutdown

htruong, 2018-05-09:
Oh if you don't want it to shut down, for now the bandaid is to set

```
NO_CONNECTION_POWEROFF_MINS=0
DISCONNECTION_POWEROFF_MINS=0
DISCONNECTION_SCREEN_POWEROFF_SECS=30
```

computeridea, 2018-05-09:
I'm not talking about the autoshutdown.. I means if i try to set the date with the command "date -s " the rasoberry istantly shutdown as it was a kind of security procedure

krizangrandi, 2018-05-09:
You don't have to create "xorg.conf.d" folder, it's already exists, there something wrong..
Have you renamed 10-evdev.conf to 45-evdev.conf?

computeridea, 2018-05-09:
No.. I didn't
Maybe this is what i'm missing
I didn't found the folder and the file.. I create by myself

computeridea, 2018-05-09:
So i will rename tjat.. And then? What will happen after it?

hitanodhav, 2018-05-10:
Hi there i managed to calibrate my screen using tslib can you please assist as to how do i create and save a file with my calibration parameters and where do i find my calibration parameters.

computeridea, 2018-05-11:
hi guys i did it, i calibrate using
this
`got it. using X11 mode and these changes

apt-get install xserver-xorg-input-evdev

cd /usr/share/X11/xorg.conf.d/
mv 10-evdev.conf 45-evdev.conf

vi 99-configuration.conf

Section "InputClass"
Identifier "calibration"
MatchProduct "eGalax Inc. USB TouchController"
Option "InvertX" "0"
Option "InvertY" "1"
EndSection`
and this

`sudo apt-get install libx11-dev libxext-dev libxi-dev x11proto-input-dev
Download xinput_calibrator somewhere in the Raspberry PI's folder structure.

wget http://github.com/downloads/tias/xinput_calibrator/xinput_calibrator-0.7.5.tar.gz
Unpack it and then navigate to the unpacked folder and then install it using:

./configure
make
sudo make install
After this step you should run xinput_calibrator(from Xserver terminal console: first startx then open console and then run it).

xinput_calibrator
Follow the on screen instructions(touching some points on screen) and after calibration is complete you will receive a message like this:

Calibrating EVDEV driver for "eGalax Inc. USB TouchController" id=8
current calibration values (from XInput): min_x=1938, max_x=114 and min_y=1745, max_y=341

Doing dynamic recalibration:
Setting new calibration data: 121, 1917, 317, 1741

**--> Making the calibration permanent <--
copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf'**
Section "InputClass"
Identifier "calibration"
MatchProduct "eGalax Inc. USB TouchController"
Option "Calibration" "121 1917 317 1741"
Option "SwapAxes" "1"
when crankshaft starts it's fine.. my touch is calbrate and work correctly, but when i connect the phone, it starts android auto.. and the touch screen seem different.. its like it work only in a more little portion of the screen, start from corner left.. it work until a little bit more of the midle of the screen, so i can't use the botton menu, and upper right voice command trigger..
anyone had same issues?

billybob871, 2018-05-12:
Am having the same problem as computeridea.