opencardev/crankshaft #65: Function bar wrong

albdom, 2018-04-03:
- What phone do you have? What OS version? Xiaomi Mi6, v0.2.1

Hi, I have reinstalled several times the image version v0.2.1 and the below bar of Android Auto tasks (naviganators, phone, music ...) does not detect my pulsations, however I am sure that the touch works in this area.

htruong, 2018-04-04:
I have not heard about this before, but let's hope that someone else reports that too so I can investigate. Did you try to start it in X11 mode to see if that does the same?

krizangrandi, 2018-04-04:
List your hardware.
Try ti make a calibration of the touch.

albdom, 2018-04-04:
I have a touch screen from Makibes. I think my calibration is correct because if I do not connect my phone I can see the pointer and go through the entire area of ​​the screen. In previous versions it had worked for me.
Additional HW: usb sound card and rotary encoder for volume control.

albdom, 2018-04-07:
You can close issue, now correct with new version!

htruong, 2018-04-07:
Interesting :) I didn't remember doing anything, but that's great to hear!

closed 2018-04-07 by htruong