opencardev/crankshaft #67: Issue with custom display - 5inch HDMI LCD V2 Waveshare

stramazzo, 2018-04-08:
Hi
I'm having a little issue with my new display 5inch HDMI LCD V2 Waveshare.
Crankshaft allways start in dev mode.
I think this is because the display driver is breaking the detection of gpio21 (pin7), the one responsable for dev mode.
Could you give me some info on wher I can disable it or move it to pin 27?

Some addictional notes:

Follwing instruction on closed issue #38  I got it to work with the last crankshaft iso.
Unfortunatelly my display, unlike the one in issue #38 , uses gpios instead of usb for touch.
The pin of the raspberry used are from pin 1 to pin 26.

The system starts in dev mode and if I login and tipe autoapp everithing is working. Apart for the failsafe mode!!!

I chose this display because:
1)it has capacitive touch so I can use it with gloves.
2)It is 5 inch so it fits in to a smartphone waterproof motorbike case.
3)The touch uses gpios and not usb so it fits in to a smartphone waterproof motorbike case.

stramazzo, 2018-04-09:
Please I do not know how bit know It is working. No Idea what was going wrong. I'll do some more tests than you Van cose the issue.

stramazzo, 2018-04-09:
I found the problem. It doesn't start if the X11 jumper is in place. If not in place It starts properly.
I guess I'll keep It this way.

htruong, 2018-04-12:
Great, thanks for the report back. Have fun! I will note that the Waveshare works using the X11.

closed 2018-04-12 by htruong