opencardev/crankshaft #68: Saving audio volume

priitpi, 2018-04-09:
Thanks for the great effort of making carputer projects fun again. 

I tried out volume changing with gpio pins using latest release (0.2.2) and it works as expected. Tho I noticed that volume level is not saved when rpi shuts down (as brightness level does). Currently it's only way I can change volume and every time rpi starts audio is really loud by default. So it would be awesome if audio level would be saved on shutdown.

htruong, 2018-04-09:
I know that's a problem. It's actually not easy to save the state of the soundcard. I need to dig into it.

Maju3, 2018-04-10:
What about a Setting for "start up" volume?
I mean instead of saving the last volume Level.
Lets say you set "start up" volume to 10% then after shutdown and boot the volume is 10%.
I think this is better as saving the last Level.
What if you hear Music VERY loud?
Then you shutdown the PI and on next start it blows your ears ;)

tehniemer, 2018-04-23:
Also interested in this.

htruong, 2018-04-23:
I have found the solution for this, this hopefully will be fixed in the next release.

hawkeyexp, 2018-07-09:
Released with CS-NG

closed 2018-07-09 by hawkeyexp