opencardev/crankshaft #70: Explanation of settings

tehniemer, 2018-04-11:
Can we get some screenshots of all the settings menus and what each option does? I am having a hard time figuring out what everything means.

htruong, 2018-04-12:
Sure, that's hopefully will be added to the wiki soon. I will need some help doing that as well.

tehniemer, 2018-04-12:
Are these settings specifically Crankshaft or are they from OpenAuto?

htruong, 2018-04-12:
They are OpenAuto's. Crankshaft will have some settings in a conf file in the next version, but not now.

tehniemer, 2018-04-12:
I looked through the OpenAuto wiki and didn't see any documentation on them at all...

htruong, 2018-04-12:
Yep, no one is doing it yet. Care to help whatever you can?

tehniemer, 2018-04-12:
I could probably get something laid out and started, but I'll need screenshots from someone else. My only touchscreen is 400x234 so it's all but unreadable.

parkerlreed, 2018-04-13:
[image]
[image]
[image]
[image]
[image]
[image]

tehniemer, 2018-04-13:
Thanks, I'll work on a draft soon.

tehniemer, 2018-04-13:
I've never seen that waiting for device screen, did you compile OpenAuto yourself?

tehniemer, 2018-04-16:
Here is a first draft. I need help with the stuff in red.

[OpenAuto Settings Guide.pdf](https://github.com/htruong/crankshaft/files/1915255/OpenAuto.Settings.Guide.pdf)

tehniemer, 2018-04-23:
Anyone? Buller?

htruong, 2018-04-23:
Thank you so much, let me help you with the rest. Sorry for the late reply, I have been busy lately. 

- The OMX layer index affects only the Pi. It overrides how the video feed from Android Auto is drawn. Normally, you need it to be 1 or 2 on EGL and 0 on X. It should be set correctly automatically on Crankshaft and requires no further tweaks.
- Margin width and height set the [overscan](https://en.wikipedia.org/wiki/Overscan) for the video. On HDMI screens, this should be set to 0. On RCA video, this might need tweaking if you encounter clippings or black bars.
- Audio Channel: See [explanation on the wiki](https://github.com/htruong/crankshaft/wiki/Using-your-Bluetooth-stereo).
- Output backend: There are several ways the Pi can play the audio. RtAudio might help with audio stuttering if you have wireless Android auto, but we don't have it now, so ignore it for now.
- Buttons allows mapping of keyboard keys to action on the android auto screen. Useful for automated testing.
- Bluetooth: I haven't had time to play with it.

tehniemer, 2018-04-24:
Thanks, you've definitely had more important things to work on, so no worries. 

I've made the updates based on your input and I only have two remaining questions before I'll publish a rev. 1.

On the general tab, what does the Handedness of traffic setting actually change? The position of the icons in the action bar?

What clock is enabled or disabled with the show clock checkbox?

tehniemer, 2018-05-13:
Can anyone help with what "handedness of traffic" and "show clock" actually do?

htruong, 2018-05-15:
I am actually not sure about the clock. 

The handedness of traffic is probably for countries that drive on the left, so the driver is on the right of the head unit. It might mvoe things around to make it more convenient for the driver to see and touch things from that perspective.

tehniemer, 2018-06-01:
First draft complete. I imagine this will be a living document, so I'll update as necessary.  

[OpenAuto Settings Guide.pdf](https://github.com/opencardev/crankshaft/files/2062710/OpenAuto.Settings.Guide.pdf)

tehniemer, 2018-06-01:
I've created a page for this on the wiki

https://github.com/opencardev/crankshaft/wiki/Understanding-OpenAuto-Settings

closed 2018-06-01 by tehniemer