opencardev/crankshaft #71: Phone calls always through phone mic and speaker

stquinn, 2018-04-11:
- What phone do you have? What OS version?
- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode
- Please provide any further information that you might find helpful.

Phone: nexus 5x Android 8.1

Phone call audio and mic are through phone not pi speaker or mic. 

Google assistant works as expected. Is there a special config required to route phonecalls through crankshaft or is this a known issue?

No I have not enabled devices mode, would that make a difference here?

htruong, 2018-04-12:
That's sadly, a known limitation. You can try to emulate the hands-free bluetooth profile (HFP) [according to the OpenAuto wiki](https://github.com/f1xpl/openauto/wiki/HFP-and-A2DP-setup-for-Raspberry-PI-3). I haven't gotten time to tinker with it, but if you can make it work, I'd appreciate you spend some time to share back.

On the other hand, if your car already had bluetooth,[ you can use a workaround to get it to work](https://github.com/htruong/crankshaft/wiki/Using-your-Bluetooth-stereo).

stquinn, 2018-04-12:
Ok, thanks for your answer. If I get some time I will attempt to set it up and if it works then I will reply here with any tips or tricks I learn. 

For ease of use though,  the car Bluetooth maybe a better route for it to just work. I had been using the aux-in of my car stereo up until now, with a usb mic. I had hoped there was some way to make the crankshaft head unit appear to the phone, essentially like a wired head set.

prinzpi, 2018-04-14:
@htruong I'm also facing this limitation... Is there maybe also an workaround with a BT receiver? For example:
[BT Receiver](https://www.amazon.com/AUKEY-Bluetooth-Receiver-Wireless-Hands-free/dp/B00O8FYUJA/ref=sr_1_6?ie=UTF8&qid=1523699754&sr=8-6&keywords=AUKEY+Bluetooth)

htruong, 2018-04-14:
@prinzpi  Yup, that should work... If your phone works with the unchecked media + speech thing.

paczatkujacy, 2018-05-31:
So for summary... There is no way to let phone calls through raspberry pi using usb sound card without BT? Also when call is over the music is redirected and playing on phone speakers rather than car speakers like before phone call. Is this normal?

ganey, 2019-02-04:
Is this still an issue? 

My Google Pixel seems to randomly decide what inputs / outputs to use with my cars built in bluetooth and is quite inconsistent.

toufou54, 2019-05-31:
Same for me with the latest build ,(OnePlus 3T)

Permanently, 2019-12-26:
Merry Christmas - any workaround yet?

Sono-Gomorrha, 2021-01-02:
I've tried this today with the latest build of crankshaft and a Pixel 3a. In general same behavior as stated above. 
I also tried to connect my phone via BT to the raspberry pi 3b+ I'm running CS on. In Android Auto this did indeed push the audio via BT to the USB soundcard and also used the speakers (ear plugs in my test setup) and microphone (usb soundcard).
But the quality was really strange and bad. When talking on the other phone (the one I called from my Pixel3a) I heard that in the ear plugs (which would be my car speakers later on), but quality as basically inaudible (music and google assistant sounds fine though). When talking into the microphone on the android auto side (which would later be in the car) there was a delay of like 3 seconds or so until I heard it on the other phone (quality was good though).

I only started to play around with crankshaft today, so I can't really speak much. It does sound like more of a BT on RPi 3 issue to me, as e.g. spotify, google maps, etc. sounds fine (so everything not using BT). Still posting it here though, as this might help others. My suggestion would be to use a USB BT dongle and test with this. Unfortunately I don't have one lying around.

matt2005, 2021-11-27:
this is an issue with version pre52 so closing

closed 2021-11-27 by matt2005