opencardev/crankshaft #72: New Android Auto layout

joacliba, 2018-04-12:
I don't know if that is an easy change but it'd be nice to have the new android auto layout, with the bar on the side, notifications, music and maps have more space to be shown.

Example:
https://i.imgur.com/qzUELf9.jpg

Maju3, 2018-04-12:
The bar on the side is NOT a new layout.This is just how it looks when you open the ANDROID Auto app on your phone.
The example you have posted is Android Auto installed on a sony tablet.
When Android Auto is in projector mode then the bar is on bottom.
This is not related to crankshaft.
If you would use Android Auto with a car Stereo like pioneer or any oem car Stereo what Supports Android Auto then the bar is on bottom as well;)

joacliba, 2018-04-12:
Ops! Sorry about that! I thought that was the new interface. I think that's a better layout for our units but, if it's not the official layout ... I have nothing to say. Thanks and sorry :)

closed 2018-04-12 by joacliba