opencardev/crankshaft #74: Maps don't sync after unplug or sleep

ldehaas1612, 2018-04-14:
- What phone do you have? What OS version?
A: OnePlus 3 running android 8.0 and android auto 3.1.581054. Crankshaft 01-04 is running on rpi 2b with the official touchscreen
- Did you try to enable and run autoapp under X11 dev mode?  
A: yes, but that made no difference 

So I'm in the delivery business and have to get in and out the car every 5 minutes. So here is what I noticed: first time I started navigation (from my driver app) and plugged the phone in. The navigation resumed on the screen and all worked fine. As soon as I unplugged it and started directions to the next address and plugged the phone in the phone freezes and maps shows no directions at all.

htruong, 2018-04-17:
Interesting. Do you mind helping us trying to narrow the problem down? 

Try with another map app, for example, Waze. I wonder if Waze does the same thing.

It does look like this problem is due to Google maps' side, but if we can't confirm it, there is little hope we can report it to Google.

ldehaas1612, 2018-04-29:
First off: sorry for the long wait on the update on this. I only just had time to really test this and started struggling with getting waze to show in the first place. Long holding or double clicking the maps icon in crankshaft doesn't seem to work for me. However the maps issue still occurs on the old regular google maps with the latest version of crankshaft. The app that opened maps on the phone and prepared the destination also crashes when running android auto the second time.
I think the next step is getting waze to show and then trying it again sharing my results.
Any suggestions?

hawkeyexp, 2018-07-09:
Give CS-NG a try and report back please.

closed 2021-11-27 by github-actions[bot]