opencardev/crankshaft #75: Android Auto wont launch on the latest crankshaft version if Android Debugging is on

the-mentor, 2018-04-14:
Hi,
It took me a while to troubleshoot and figure out that with the latest version of crankshaft if i have android debugging turned on it wont launch android auto when i plug in the phone.
if i turn USB debugging off it works as expected this is a regression since it used to work with the older versions of crankshaft.

This is really important since I always leave USB debugging on.

Thank you!

htruong, 2018-04-14:
I have USB debugging on and it doesn't happen to me, but let me figure out what is going on :) Thanks for the bug report. I hope once we figure out the wireless connection, this problem is gone once and for all.

ldehaas1612, 2018-04-15:
Having the same issue. Changing the USB configuration to MTP fixes the issue. you can usually set it permanently through developer options.

knotbeerdan, 2018-04-16:
I have kind of the same issue with my moto z force, only android auto will not launch until I unlock my phone, before it would launch regardless of being locked.

My USB debugging is turned off

tehniemer, 2018-04-17:
I have noticed the same issue as @knotbeerdan

htruong, 2018-04-19:
I will investigate this bug as soon as possible.

closed 2020-07-19 by the-mentor