opencardev/crankshaft #77: Crankshaft shuts down immediately after NTP time sync

snailium, 2018-04-16:
- Enabled dev mode
- SSH'ed to Crankshaft
- Setup NTP to sync time (time jumped from 2016 to 2018)
- Crankshaft shuts down immediately after boot/reboot

I guess problem is the auto-shutdown script in the latest build.

htruong, 2018-04-16:
Ahaha! Oh yeah, do a `shutdown -c` before you sync the time :) When it starts up, it sets a 15 minutes shutdown in the future just in case something starts up the Pi and doesn't shut it down (it will drain the car battery). When you plug in the phone, it cancels the shutdown timer. :)

htruong, 2018-04-16:
So... tag this as `wontfix`?

htruong, 2018-04-16:
Oh I see. So ntp syncs the time after the timer was set on every start up. This isn't nice. I guess we can somehow make it so the crankshaft startup script waits for ntp to finish before it sets the timer.

snailium, 2018-04-16:
Maybe it is better to disable auto-shutoff in dev mode?

htruong, 2018-04-19:
Done. Edit the `/boot/crankshaft/crankshaft_env.sh` and set that `NO_CONNECTION_POWEROFF_MINS` to `0`, and it won't auto-shutoff.

closed 2018-04-19 by htruong