opencardev/crankshaft #78: Issue connecting phone

wtippet, 2018-04-17:
-Galaxy S9 / Android 7.0 Nuget
-running crankshaft 04-07-2018 build on rpi3
I can boot up crankshaft into x11 for a custom monitor but for some reason I can't get past the connect phone screen. after 5 mins or so the pi goes into a screen saver state and the monitor loses connection.
Is there a setting on my phone that I need to set or is there a setting in the program I need to set. 
I read another post that suggested turning off the usb debug mode so I did that but no luck.  
I also checked the working equipment posts but there was no mention of the software working with my particular phone.

wtippet, 2018-04-17:
disregard I found the problem. It appears that the software/pi are very touchy with power. I found that I was able to get the program to work by upgrading the power supply connected to the pi. It works great now. -with debugging on I might add.

htruong, 2018-04-17:
Great to hear, 

Thanks for the report! I am closing since the issue is resolved, but the issue with power was dutifully noted.

closed 2018-04-17 by htruong