opencardev/crankshaft #79: Question: Does it work with a Pine ROCK64?

Hoerli1337, 2018-04-21:
I ordered a Pine ROCK64 (Not here yet) and would like to ask briefly whether Crankshaft also works with the Pine.
I have two smartphones here I could test it with.
A Samsung Galaxy Note 8 Duos and a Note 3 (with Custome Rom).
If it doesn't work, I would be very happy if it is supported soon.

htruong, 2018-04-21:
Probably not in the immediate future. I think next board I'm going to support is the TinkerBoard. But even that board is going to be a couple months out. 

All those boards do have a chance, it just takes money and time to investigate. I'm just an independent developer with not many resources. So patience is key.

Hoerli1337, 2018-04-22:
I bought the ROCK64 because it has much more power and the pi seems to be a bit overloaded (when I watch the videos on YouTube).
I am aware that it takes a lot of time and money and that you are working on it alone.

TomK, 2018-04-23:
Also interested in the rock64.  I'll poke around the code and try to learn more about the process involved with adding support for other boards.  @htruong Any (even brief) process you may already have for this would be really appreciated.  👍

Hoerli1337, 2018-04-23:
@TomK You're my man!
If you need a test person, I am at your disposal.
I can't program, but find errors :)

htruong, 2018-04-23:
@TomK: 
If you need to get it to work with any custom board, I think the general direction is generally this:
- Start with a clean stock image from the provider: raspbian, rock64 image, tinkerboard...
- Try to compile Qt with the script in `prebuilt`.
- Try to compile OpenAuto & aasdk according to aadsk wiki. Try to see if that works.
- If it does, you need to "rescue" those binaries out of your test machine.
- Transplant the binaries to another clean image (so we only have binaries, but no source) to see if that still works.

If all those works, you'll have to begin modifying `crankshaft.sh` to ask it to make it work with your board of choice. It does three things basically:

- First, it downloads and resizes the image so we have enough space for the stuff we're going to install. It also copies the binaries to the image.
- Second, it `chroot`s to the image and `apt install` more stuff to the image by executing `customize-image-pi.sh`.
- Third, it does some tweakings to make the filesystem read-only by executing `read-only-fs.sh`

To make it work minimally, you only need to do 1 & 2. 3 is the bonus, but I think it's not necessary.

htruong, 2018-04-23:
@Hoerli1337 what a pity: All the performance issues have been resolved in the last month. The Pi works just fine now.

Hoerli1337, 2018-04-24:
@htruong I got now a Pi 3 B+ and install it.
Run perfect. Tested with Note 3 (Custom ROM) and Note 8 Duos (Stock).
I saw a view videos which are older than 2 month and they have some laggs.
I wanted to avoid this by the stronger performance of the ROCK64.

I realy like your work and hope that you will continue to support it.
I also hope that @TomK gets it to run on the ROCK64.

TomK, 2018-04-25:
Just thought I'd add that I have run crankshaft on my Pi 3B with pretty good performance.  So even without the rock64 i'm happy.
My interest in the rock64 is because i took part in the initial pine64 kickstarter, and also because the performance is supposedly fantastic.

htruong, 2018-04-25:
Great to hear. We will eventually expand to support other boards. But for now, it's a good thing to keep everything focused until we have more developers :)

HahnRobin, 2018-05-03:
Hi! 
I tried it (only with OpenAuto, but it is part of Crankshaft) and it is working. Slow, but working. I haven't optimized anything, just compiled it. I tried it with a Huawei P10 and I can use the maps and Spotify and everything. 
I think, it is possible to get Crankshaft running on the Rock64. I will try to test Crankshaft when I will have a bit more time :) Keep up the good work!

j-ak-e, 2018-06-15:
I'm interested in supporting these boards, as they're significantly lower cost, and have marginally lower power overheads than the Pi, it's good news that OpenAuto compiles and runs, that means Crankshaft _can_ be supported, however I'll have to wait until I have hardware to test with

inspector71, 2018-08-21:
I think GPU drivers are the key to any of these SBC boards, rather than fabulous-sounding hardware specs. The VideoCore IV GPU on the Pi [has received paid open source driver development](http://www.anholt.net/) after [Broadcom opened up some of the source code in Feb 2014](https://www.broadcom.com/blog/android-for-all-broadcom-gives-developers-keys-to-the-videocore). 

FWIW, I have a ROCK64 I've hardly used since purchasing it. AFAIK Pine rely on community builds to make their boards usable. That's not ideal. Who (or what) is "ayufan"? Turns out, it's a guy from Poland. His builds may work fine but they may not. 

Most of [these software images](http://wiki.pine64.org/index.php/Rock64#ROCK64_Software_Images) do not appear to have received updates for 12 months. Also on e year ago, a release engineer for Fedora stated 

> biggest holdout is MALI, and that would need two open drivers or ARM to come to the table

in his _[State of open source accelerated GPUs on ARM article](https://nullr0ute.com/2017/09/the-state-of-open-source-accelerated-graphics-on-arm-devices/)_.

The ROCK64 sports an [ARM MALI GPU](https://developer.arm.com/products/graphics-and-multimedia/mali-gpus/mali-450-gpu) that may be _**theoretically capable**_ of 4K@60 FPS but perhaps that is like saying your car is capable of enormous power and top speed ... you'll rarely be able to experience that ... certainly on public roads, legally!

@HahnRobin well done getting OpenAuto compiled on ROCK64! Is GPU acceleration enabled?

closed 2025-01-12 by matt2005